Nickduino/Pi-Somfy #1: Getting started

waltner, 2016-10-03:
Hi,

I just found your repo and would be interested in some more details:
1) How to set chambre.txt values?
2) What kind of remote you are using, I have a "Somfy Roma" with three Buttons.
3) What emitter you use, I have a cheap 433 MHz emitter from eBay.

Merci beaucoup in advance for your help,
Georg.

Nickduino, 2016-10-03:
Hi,

1. "chambre" is the name of the remote. You can set as many remotes as you'd like.
      - First line is the address. Choose a random Hex value, 0x267043 in my example.
      - Second line is the rolling code. It will be incremented by the script every time the stores are moved up or down;
2. I have one with three apparent buttons **plus** the hidden "Prog" button;
3. Any cheap eBay emitter will work __if you replace the crystal for a 433.42 MHz one__.

RaspiGuru, 2017-03-08:
Hi,
It would be very helpful to me if you explain that how can I add/delare my raspberry (virtual) remote at the first time to my blind?
Thank in advance
With regards
Raphy

closed 2017-03-08 by Nickduino