Nickduino/Pi-Somfy #2: Can you help with a supported transmitter ?

Strudsen, 2017-01-03:
I really like this project, but I cant get the pi to communicate with the blinds, I am new to this kind of stuff, but i am pretty sure that my problem is the transmitter.

I have bought this one:
http://www.ebay.com/itm/5x-433MHz-RF-Transmitter-Receiver-Kit-Module-Arduino-ARM-WL-MCU-Raspberry-PI-/162102463039?hash=item25be0f4a3f:g:xikAAOSwnNBXXm2E

and replaced the Crystal with this one:
http://www.ebay.com/itm/331637441887

but they don't respond at all

I have a working Telis 6 and Telis 5 remote for the system today

Could it be possible for me to buy a working transmitter module for the raspberry pi from you ?

Nickduino, 2017-01-09:
Hi,

Excuse me for the delay.

It seems you're using the same transmitter and crystal than me.
Are you sure you're using the script properly? How did you register the emulated remote with the blinds?

Strudsen, 2017-01-10:
Thanks for you reply.

I got it work, is was a antenna problem, after I replace the antenna wire it start to work just fine.

again thank you

Nickduino, 2017-01-11:
Excellent!

Care to tell me more about your install?
How many blinds? How smart have you made them?

RaspiGuru, 2017-03-08:
Hello,
I'm interested to know that how did you register the emulated remote with the blinds?
Thanks for your help.
With reagards
Raphy

Nickduino, 2017-03-08:
Long press (until the blinds move) on the "Program" button of your original remote. Then send a frame with the program value (0x8) from your emulated remote.

If you find it any easier, you can do it with your Arduino.
https://github.com/Nickduino/Pi-Somfy/blob/master/somfy.py

closed 2017-03-08 by Nickduino

RaspiGuru, 2017-03-17:
Hello,  
I tried multiple times to send a frame as you describe on your mail but no succes.   
  
here is the command sent and the results:  
  
0x267043  
133  
Remote  :      0x267043  
Button  :      0x08  
Rolling code : 133  
  
Frame  :     0xA7 0x80 0x00 0x85 0x26 0x70 0x43   
With cks  :  0xA7 0x8C 0x00 0x85 0x26 0x70 0x43   
Obfuscated : 0xA7 0x2B 0x2B 0xAE 0x88 0xF8 0xBB   
  
Have you any idea please?  
Thnks in advance for your replay.  
Raphy  
  
Long press (until the blinds move) on the "Program" button of your original remote. Then send a frame with the program value (0x8) from your emulated remote.

  If you find it any easier, you can do it with your Arduino.
https://github.com/Nickduino/Pi-Somfy/blob/master/somfy.py

    �

Nickduino, 2017-03-17:
Do the blinds move after the long press on the original remote?

Are you sure you're transmitting on the right frequency?

RaspiGuru, 2017-03-17:
Hi,  
Yes the blinds move after the long press but I'm not sure of my tramitting frequencies. Please see the attached image for my two types of 433 Mhz transmitters I'm using.  
I can control my electrical plugs without any problem but they are responding to fixed codes.  
Thnaks for your any advices  

With regards  
Raphy

Nickduino, 2017-03-18:
I can't see your attachements (if you sent them by mail, they won't appear on github) but  your transmitters are the most likely culprits. You have to transmit on 433.**42** MHz, not 433.92 (by far the most commmon frequency). So, you can:

- hack a remote and send stuff through it;
- replace the SAW oscillator of your transmitters for one at the right frequency (I found some on eBay);
- use a tunable transmitter (if you can find some ham radios guys to help you, they should prove useful)

Cheers,
   Nick

RaspiGuru, 2017-03-18:
Hi thanks again,
I've ordered (at http://www.ebay.com/itm/331637441887) a 433.42 Mhz Crystals on eBay.
And will back to you if necessary. 
Great thanks
With regards
Raphy

upiw, 2018-09-11:
@Nickduino - I was wondering if it might be straight-forward to modify your project to work with Simu roller shutters.  I've read somewhere that they also use 433.42 Mhz.  But would the signal protocols be different or is the Somfy one using a standard protocol?

Nickduino, 2018-09-11:
That would be pretty easy: you don't have to change anything.

The protocol isn't standard at all but Somfy bought Simu a long time ago and made them use their RTS protocol (naming it "Hz"). In fact, I have Simu blinds myself...

upiw, 2018-09-11:
Thanks so much @Nickduino.  I have some 433.42 mHz crystals on the way.
I tried to use the project to control a Somfy using a Jaycar 433.92 transmitter but had no response with that.  I'm kinda doing similar to @RaspiGuru.  See how it goes.  Thanks again for the excellent project.

Nickduino, 2018-09-11:
It CANNOT work with at 433.92 MHz. The receivers are incredibly picky

mjasny, 2018-10-12:
> That would be pretty easy: you don't have to change anything (the Readme even mentions it).
> 
> The protocol isn't standard at all but Somfy bought Simu a long time ago and made them use their RTS protocol (naming it "Hz"). In fact, I have Simu blinds myself...

Are you able to control your Simu blinds? I have a SIMU FR-70103 Gray EMETTEUR Hz 1C (Channel).

I'm can record signals with my Arduino and analyze them, but the protocol does not seem to fit. I also tested your ready-to-go script but I was not able to register my arduino-remote.

Nickduino, 2018-10-12:
I believe the "Hz" you mention is the Simu Hz protocol, the same as Somfy RTS.

So, yeah, it should work. It's what works for me (my blinds come from the Simu).

If you see the frames with your Arduino sniffer, I suppose your wiring is good. What procedure do you follow to register your new virtual remote?

mjasny, 2018-10-12:
The blinds, including the remotes were installed one month ago, I hope 
that they did not change the protocol.

I hold the 'Prog' Button on the existing remote until the blind moves 
up-down. Then I press 'p' in the serial-monitor. I also tried different 
Hex-Codes for other Keys with no success.
After that, my existing remote does not work until I press 'Prog' again 
following one of the other keys.

Nickduino, 2018-10-12:
Your mixing up the Arduino sketch and the Python script and putting all that into an old issue.

You might want to get your facts straight first.