Nickduino/Pi-Somfy #3: Open-source license?

benesch, 2017-04-11:
Hi @Nickduino, thanks so much for making this code public! Just wanted to ask if you'd be willing to throw some sort of a open-source license on the code—I'm hoping to publish a modification that integrates with Amazon Alexa, and I don't want to be stealing your work. :beers:

Nickduino, 2017-04-11:
Hi benesch,

I'm glad the code is useful to some people! I myself based it on a protocol decoding found on Pushtack.

Maybe you could fork the repo? Or just mention me in your code?
I don't want to bother people with licensing: use the code and be happy :)

closed 2017-04-11 by Nickduino

tholu, 2017-05-23:
@benesch Have you started on working on a version that integrates with Alexa? I'm planning the same thing.

benesch, 2017-09-25:
Afraid I haven't made any progress yet.