Nickduino/Pi-Somfy #4: Update README with Getting Started

tholu, 2017-05-23:
Great project! I want to use that as a starting point for my Somfy blinds as well.

Can you add a "Getting Started" section to the README, that explains for beginners how to get this basically working? I already have a transmitter and the custom crystals, and need to figure out the following:

- Connect transmitter with replaced crystal to Raspberry Pi
- Register transmitter as additional Remote for each blind
- Actually control the different blinds, to see if everything is working

What I want to add then is:

- Alexa integration

Nickduino, 2017-05-23:
Hi, 

Thanks for your input.

I should totally improve the readme! Procrastination, procrastination...

__In short:__
Power your transmitter with the Pi and connect its input to the pin you choose as TXGPIO (default is GPIO 4). I do power **mine** with the 5V rail and it works flawlessly with the 3.3V Pi's output but I can't guarantee it will be the same for you.

To register  a new remote, you have to press the Program button of your original remote (generally, you have to remove the plastic cover) and hold it until your blinds move up a bit  and down a bit, then send a program command from your virtual remote (the code is 0x8, you can either modify the Python script to do so or use the [arduino sketch](https://github.com/Nickduino/Somfy_Remote)).

You can register the same virtual remote to several blinds if you wish but do them one by one.

Let me know if you have any problem, we'll try to troubleshoot them. And kudos for the Alexa integration, that will be sweet.

tholu, 2017-05-24:
@Nickduino Thanks for your fast response, I will try that (perhaps I can update the README already while doing it)!

Nickduino, 2017-05-24:
Sure, try so do a pull request or something with an improved readme 😊

steph-dam, 2017-09-06:
Hi,
First, many thanks for your code. I test it on my pi without any modification and while using your example files, I have an error occuring : _"TypeError : bytearray indices must be integers or slices, not float"_ corresponding to the code : _if ((frame[i/8] >> (7-(i%8))) & 1):_
Any idea of what happens ?
Thx

Nickduino, 2017-09-06:
That's weird. What happens if you print your i value?

steph-dam, 2017-09-09:
if I print i just before the line 'if ((frame...' it returns 0.
It don't make error if I use '//' instead of '/'  in frame[i/8]. But is it correct in the logic of the program ?

Nickduino, 2017-09-10:
And after the division?
// marks a comment in C/Arduino, so there is no chance the code will work.

closed 2017-09-10 by Nickduino

steph-dam, 2017-09-11:
In fact, as I said in my first question, I have a raspberry pi so I've used the python download (Pi-Somfy).
As i is increasing from 0 to 56, i/8 can't be an integer. That's why I've tried the integer division. But I'm still waiting for the SAW resonator at 433.42 MHz so I can't make trials with my blinds.

Nickduino, 2017-09-11:
Sorry for the comment about comments, I mixed things up.

i/8 for i=1 should be 0.
And i%8 for i=1 is 1. It the remainder of the Euclidean division.
And i%8 for i=19 is 3. Still the remainder of the Euclidean division.

But Python **3** behave differently (thus why many people prefer to use Python 2.7) than other languages. **So, in short, yes for i//8.**

steph-dam, 2017-09-12:
Thanks to have confirmed and you are totally apologized for your comment  :)
If it works, I will put a new comment and also try the code in python 2.7