Nickduino/Pi-Somfy #5: What kind of emitter to use?

z4nD4R, 2017-08-01:
Hello,

such a great articel. I'd like to use what kind off emitter is "best option" to use?

Thank you.

Y

Nickduino, 2017-08-01:
Hi,

Any cheap transmitter should work, the cheapest I've tested with success is this kind: http://www.ebay.com/itm/5x-433MHz-RF-Transmitter-Receiver-Kit-Module-Arduino-ARM-WL-MCU-Raspberry-PI-/162102463039?hash=item25be0f4a3f:g:xikAAOSwnNBXXm2E
(You just need one, not 5)

Make sure your replace the transmitter for the proper frequency, though!
http://www.ebay.com/itm/331637441887

closed 2017-08-01 by Nickduino

waltner, 2017-09-13:
@Nickduino Are you sure it is necessary to replace the transmitter? Somewhere I read it should be fine also with 433.92 Hz, only the range should be limited to a few m. (Which might be enough for some users).

z4nD4R, 2017-09-13:
@waltner I've tried with given lib and default transmitter and it didn't work. Afaik the `physically coding` of SOMFY protcol allows to use little bit different freq. but still didn't work.

Which remained me, that my crystal still didn't arrived.

Y

Nickduino, 2017-09-13:
During my testing, I had absolutely no luck with the original 433.92 MHz SAW oscillator. Other users have had the same result.

But nothing prevents you from trying.

waltner, 2017-10-17:
I can finally confirm that also for me it only works with the replaced transmitter. ;)

Nickduino, 2017-10-17:
Yeah, that's astonishing how picky they are. I guess that makes their job easier, as the 433.42 MHz band is much less polluted.

alxlaxv, 2018-02-27:
Hello guys,

I got my hardware now, can you please help to plug it?
I have this: https://www.ebay.ch/itm/272993898467?ViewItem=&item=272993898467

And I will change the crystal with this: https://www.ebay.ch/itm/232574365405?ViewItem=&item=232574365405

Thanks for your help

Nickduino, 2018-02-27:
I reckon it will work.

I personally used GPIO 4 but you can change this line to anything you want:
`TXGPIO=4 # 433.42 MHz emitter on GPIO 4`

Switzerland: I took a microcontroller class from the EPFL on Coursera :)

alxlaxv, 2018-02-28:
I’m pretty basic in electronics so could you detail all wires I need to use
and where to plug it?
I guess there is power also and which transmitter pin to use ?

Oh cool! I’m not that far, Switzerland too. If ok we could have a quick
chat over emails if you want.

waltner, 2018-02-28:
There are three pins on the transmitter, you need to connect all of them: ATAD (DATA), VCC, GND
As @Nickduino mentioned, connect the ATAD pin with GPIO 4 (or adapt for your needs), VCC with 5V/3.3V (not sure if 3.3V is enough?) and GND with any GND pin on the RPi.

Nickduino, 2018-02-28:
> Edit: See this [comment](https://github.com/Nickduino/Pi-Somfy/issues/4#issuecomment-303398690), 3.3V should work.

What I was saying, is that I __power__ the emitter with the 5V rail and that the 3.3V __data__ output of the RPi does trigger it. But 3.3V power might work, I didn't bother trying.

waltner, 2018-02-28:
> the 3.3V data output of the RPi does trigger it
Ok, my bad. Was not obvious to me when I skimmed over it.

Nickduino, 2018-02-28:
Most of us aren't natives, so we ought to encounter misunderstandings from time to time.

alxlaxv, 2018-02-28:
Thanks for your feedback!
Everything is plugged now but I don't really get how the remote.txt file and the python script have to be runned...
What should I put in the txt file?
And for the py script? Basically you seam to have several shaders, I just have one. 
How should I do?

PS: I can see a lot of french inside, could we have a quick chat over gmail or irc?
Thanks