Nickduino/Pi-Somfy #6: Where can I buy the emitter ?

grout-r, 2018-02-27:
Hi,

Do you have any recommendation about which emitter buy ?

I looked on Aliexpress but I don't know what to buy.

Thanks

grout-r, 2018-02-27:
Oops didnt look in closed issues, you can delete this one

closed 2018-02-27 by grout-r

Nickduino, 2018-02-27:
> Oops didnt look in closed issues, you can delete this one

Kudos for searching by yourself! Let me know if tou run into issues. If you are into Arduino as well, you can check my other repo ;-)

grout-r, 2018-02-27:
Ok thanks !
As i like python better than C, i'll stick with raspi for the moment :)

Nickduino, 2018-02-27:
I wouldn't have been able to do it in C on the Raspberry Pi (thank god I found that C library you can use with Python) but Arduino is easy.

z4nD4R, 2018-02-27:
@Nickduino I'm just finghting with somfy and ESP32 do you have any useful tutorial?