Nickduino/Pi-Somfy #7: Question: Will it work with DC306 remote?

szuniverse, 2018-03-02:
Hi.

I have almost the same remote controller, this one:
https://sukcesgroup.pl/wordpress/wp-content/uploads/2014/11/ARTISTIC_1_5_15R_ENG.pdf

What do you think, will it work with your sollution? Or this remote from a totally different brand with different remote implementation?

thanks in advance

Nickduino, 2018-03-02:
It's not the right frequency (433.42 vs 433.92).

With my solution, you don't need a remote, just an emitter: the scrypt _is_ the remote.

closed 2018-03-02 by Nickduino

szuniverse, 2018-03-02:
Thanks your fast reply! 
in the documentation: Emission frequency 433.92Mhz +/- 100kHz (top left corner on the doc sheet)
So i think the freq is the same. 

Another question:
I have a master remote controll which means only this device can teach other remotes. How can I put the script onto learning mode? Or how can I configure this script? This is the only part which is not so clear for me.

Nickduino, 2018-03-02:
It isn't the right frequency.

The remotes don't learn anything: the blinds do.
Put your blind into learning mode with its own remote (not the master remote) and then send 0x8 with the script to register the new virtual remote.

szuniverse, 2018-03-02:
Okey I understand how to register new device, but why is not the same freq?
Your script works 433.42 vs 433.92 and the doc sheet said the emission frequency 433.92Mhz, where is the difference?

Nickduino, 2018-03-02:
Somfy blinds want 433.42.

Your remote is 433.92.

szuniverse, 2018-03-02:
Okey but in this case I only should use 433.92 emitter on the raspberry side (remote). am i right?

Nickduino, 2018-03-02:
You need a 433.**42** MHz emitter. Personally, I bought a chinese 433.92 one for $1 and soldered a new SAW oscillator.

szuniverse, 2018-03-02:
Okey, Thanks your help!

Nickduino, 2018-03-02:
Glad my work is useful somehow.