Nickduino/Pi-Somfy #8: Make it work

alxlaxv, 2018-03-04:
Hello,

I'm trying the code provided but nothing seems happening.
Python is pretty new to me.
Do I have to register the remote? Can you please provide some code examples?

Thanks!

Nickduino, 2018-03-04:
You have to register your new virtual remote, yes.

Long-press the prog button of the blind's remote to make your blind enter programming mode. Then send command 0x8 with a python scrypt to register the new remote.

alxlaxv, 2018-03-04:
Could please give a code example to do that? 
Thanks you so much

Nickduino, 2018-03-04:
I haven't done it in years :D

Last line:
`envoi_commande("your_remote_name", 0x8)`

alxlaxv, 2018-03-04:
1. Long-press the prog button of the blind's remote to make your blind enter programming mode.
2. Blinders go down a bit and then up a bit
3. I send the reg
```
sudo ./reg.py
0x267043
134
Remote  :      0x267043
Button  :      0x08
Rolling code : 134

Frame  :     0xA7 0x80 0x00 0x86 0x26 0x70 0x43
With cks  :  0xA7 0x8F 0x00 0x86 0x26 0x70 0x43
Obfuscated : 0xA7 0x28 0x28 0xAE 0x88 0xF8 0xBB
```
4. I send the closing instructions
```
sudo ./som.py
0x267043
135
Remote  :      0x267043
Button  :      0x04
Rolling code : 135

Frame  :     0xA7 0x40 0x00 0x87 0x26 0x70 0x43
With cks  :  0xA7 0x42 0x00 0x87 0x26 0x70 0x43
Obfuscated : 0xA7 0xE5 0xE5 0x62 0x44 0x34 0x77
```

5. Nothing happens. Few seconds later, I got a small move probably, Prog mode turning off.

How can I be sure the query is well sent?

Nickduino, 2018-03-04:
Looks ok at first glance.

Did you triple check your wiring?

Sure the emitter is broadcasting on the right frequency?

alxlaxv, 2018-03-04:
I changed the crystal today with the one mentionned in the past.
Is there a way to be sure that the transmitter is well detected? plugged?

Nickduino, 2018-03-04:
> Python is pretty new to me.

Are you more comfortable with C (Arduino)?

alxlaxv, 2018-03-05:
I only have a RaspberryPi and python is closer from what I know :) 
So

1. I have triple check plug
-     Connect the ATAD pin with GPIO 4 
-     VCC with 5V
-     GND with any GND pin on the Raspberry Pi
based on https://www.raspberrypi.org/documentation/usage/gpio/

2. I have check wires, working

3. I have check power with a small plugin in parallel on the emitter

Is there a way to be sure that the transmitter is well detected? plugged?

Thanks!

Nickduino, 2018-03-05:
99% sure it's the wiring.

Pigpiod uses the **Broadcom** (BCM) numbering. The link you posted shows the **board** numbering.
If you're new to the Pi, it's confusing.

I'll let you check __your__ BCM GPIO (they've changed with the Pi's revisions) and I'm confident it will work once connected to the right pin.

Alternatively, you could use an oscilloscope to see on which pin data is transmitted but that's quite overkill :-D
Maybe a decent multimeter would work as well.

alxlaxv, 2018-03-05:
It seems not as I'm plugged on the number 4 on my link... which is BCM pin 4 based on that link https://pinout.xyz/pinout/pin7_gpio4#

What do you think?

Nickduino, 2018-03-05:
I think you should write a program lightening up GPIO4 (or any, really) to make sure you identified the right pin.
Or maybe, just put an LED on GPIO4 with my Python script, you might see it flicker (not sure it will light up long enough, though).

If it's the right pin, try powering your emitter with 3.3V instead of 5V?

Check your soldering.

Try another emitter? Another crystal?

alxlaxv, 2018-03-06:
Update, I have made the same setup but with a LED and this code:
```
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(4,GPIO.OUT)
print "LED on"
GPIO.output(4,GPIO.HIGH)
time.sleep(2)
print "LED off"
GPIO.output(4,GPIO.LOW)
```
And it's working fine.
I will have to look the emitter closer or the crystal...

Nickduino, 2018-03-06:
That's probably the problem.

Did you try powering it with 3.3V instead of 5V?

alxlaxv, 2018-03-07:
Yes I tried both.
I plugged my leds in parallel, they are blinking 3 times when I execute the script but nothing happens

alxlaxv, 2018-04-03:
I have made another emmiter which seems to work.
Always order chinese products by 5, not 1.

Nickduino, 2018-04-03:
Excellent! I'm happy for you.