Nickduino/Pi-Somfy #9: Timing and Frame repetition

GKFR, 2018-03-23:
Hello,

I used your code and succeeded in registering and controlling my Somfy shades using an Aurel Transceiver (433.42MHz). I also checked the PushStack blog with details about the RTS protocol your code is based on. I noticed that you use a different timing (640us per half bit instead of 604us documented in the PushStack blog). The PushStack blog mentions that the timing is not critical but I wonder if there a special reason for that? 
I have a few open questions, maybe you can help me:
- Is it necessary to repeat the frame 3 times as you do in your code?
- Do you have an idea how to implement a long press of the prog button to register/de-register a remote? Is this established by often repetition of the frame?
- My original Somfy remote has a separate wheel (additionally to the 3+1 buttons) to adjust the angle of the blinds. My Somfy Telis 6 Chronis remote does not have this wheel but the angle can be adjusted by long pressing up or down. Do you have an idea how to implement this function? Is this a separate control code or is this established through frame repetition?

Thank you in advance!

Nickduino, 2018-03-23:
Firstly, kudos for your success!

Then:
- 640µs vs 604µs: 604µs is a typo. You can check the patent (I quoted it in the closed issues either here on on my Arduino repo);
- frame repeat: since the emitter doesn't have any way of knowing whether or not the receiver got the signal, it's safer to send it multiple times. Even your weather station does it this way;
- no clue. If those buttons aren't mentioned somewhere on the internet (PushStack gives you values from 0x1 to 0xA), I'm afraid you'll have to reverse-engineer them either with an oscilloscope or by building a Somfy decoder.

GKFR, 2018-03-23:
Thank you for your quick response and the helpful comments. I will try to find and check the patent and play with my transceiver to be able to receive signals from the Somfy remotes I have available. Let’s see what I can find out.

GKFR, 2018-03-30:
In a first attempt to answer my questions I played around with my program and could finally answer all my questions.
- As you said, it is not necessary to repeat the frames at all. My blinds respond to single frames already. However, as you said, repetition of the frames raises the chances that at least one of the frames is received if there is interference.
- I tried to send more repetitions of the same pulses to simulate the tilting function for the blinds. Sending more than two repetitions after the original frame means a button kept pressed and moves the blind in steps to adjust the tilt. Sending the original frame and three repetitions is the smallest adjustment, sending the original frame and more repetitions moves the blinds up/down for a longer time.
- For the prog button the principle is the same. To activate the register (as with a long press on the prog button) you have to repeat the frame. I tried some repetitions and it seems that it is necessary to repeat the frame at least thirteen times after the original frame to activate the registration.

Thank you for your help and the motivation!

Nickduino, 2018-03-30:
> Thank you for your help and the motivation!

Well, that was a real pleasure to discuss these matters with a smart and concise fellow enthusiast.
If you want to include your discoveries in Pi-Somfy, do a pull request.

closed 2018-03-30 by Nickduino

paulc1983, 2022-06-25:
Hi 

Firstly thanks for this project!

I know this is an old discussion, but i thought i would ask here first as its related to what i am trying to achieve.

Did this ever get merged in? Im trying to find a way to tilt my blinds which is done with  a long press on the physical remote.  i have tried this but nothing happens :
sudo python3 /home/pi/Pi-Somfy/operateShutters.py "Left blind" -c /home/pi/Pi-Somfy/operateShutters.conf -press btn up -long

have i got the cli wrong? 

Is  there w way to simulate a long press and specify the length of time the button is held down? I want to see if i can find a way to do this in the manual operation  with mousedown and mouseup events and maybe cancellation tokens or repeated calls to the button. Im not sure if it is possible though?