Nickduino/Pi-Somfy #11: IndentationError in somfy.py

waras2017, 2018-04-28:
Hello,

thank you for your code example to show how one can control somfy blinds.

I just tried to register my raspberry as a new remote. But when I eecute the somfy.py I get an IndentationError in the sending function.

It seems to me that the problem is in that area:

---
.
.
.
   wf.append(pigpio.pulse(0, 1<<TXGPIO, 30415)) # interframe gap

      for j in range(1,repetition): # repeating frames
            for i in range(7): # hardware synchronization
                  wf.append(pigpio.pulse(1<<TXGPIO, 0, 2560))
                  wf.append(pigpio.pulse(0, 1<<TXGPIO, 2560))
            wf.append(pigpio.pulse(1<<TXGPIO, 0, 4550)) # software synchronization
            wf.append(pigpio.pulse(0, 1<<TXGPIO,  640))

            for i in range (0, 56): # manchester enconding of payload data
                  if ((frame[i/8] >> (7 - (i%8))) & 1):
                  wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))
                  wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
                  else:
                  wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
                  wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))

            wf.append(pigpio.pulse(0, 1<<TXGPIO, 30415)) # interframe gap

   pi.wave_add_generic(wf)
.
.
.
---
The number of blanks in front of the "for j in range(1,repetition)... could be the problem.

Could you please check the code and/or give me a hint how to use it in a correct way.

Thanks a lot!
Uwe

Nickduino, 2018-04-28:
Are you using Python 2.7?

What did you modify in the code?

waras2017, 2018-04-28:
Yes python 2.7

I took your code and removed everything below the sending function and put only
one Prog Sendin Command in.

Meanwhile I reorganized the lines above in the sending function and now it worked for me!

Nickduino, 2018-04-28:
Perfect, then.

closed 2018-04-28 by Nickduino

Nickduino, 2018-09-30:
> Meanwhile I reorganized the lines above in the sending function and now it worked for me!
Did you indent the four _wf.append()_? Thinking I should update the code but I can't test it on my mobile...