Nickduino/Pi-Somfy #12: some problems

turuwhiska, 2018-09-29:
hi. I had a problem with somfy.pi:
1) indentation Error in the sending function: this solves it by deleting all that
for j in range(1,repetition): # repeating frames
        for i in range(7): # hardware synchronization
              wf.append(pigpio.pulse(1<<TXGPIO, 0, 2560))
              wf.append(pigpio.pulse(0, 1<<TXGPIO, 2560))
        wf.append(pigpio.pulse(1<<TXGPIO, 0, 4550)) # software synchronization
        wf.append(pigpio.pulse(0, 1<<TXGPIO,  640))

        for i in range (0, 56): # manchester enconding of payload data
              if ((frame[i/8] >> (7 - (i%8))) & 1):
              wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))
              wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
              else:
              wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
              wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))

        wf.append(pigpio.pulse(0, 1<<TXGPIO, 30415)) # interframe gap
The question is that I do not know what is affecting the operation.
2) when it starts, it only shows me: Sunset: 2018-09-29 17: 25: 13.000005 I simply execute the command, I do not know if it is necessary to add extra parameters for press the virtual butons? (./somfy.py ????)
2) Does this soft serve to make snnifer? Where do I get the data from the already programmed remote control? Can I use it as a new control and make a new pair?
3) I have another system niceforyou mark that supposedly works at 433.92 this soft serve to simulate the rolling code at that frequency?
thank you very much

Nickduino, 2018-09-29:
1. You removed the most important part
2) I'm only giving you a blank canvas, **you** have to program it to do whatever you want (my example is closing at night and opening every morning but you could think of anything)
2) (again !?) It doesn't sniff anything, it's not even connected to an RF receiver. It will emulate a new remote
3) Nope

turuwhiska, 2018-09-29:
thanks for you response.how can resolv the problem:
  File "./somfy.py", line 115
    for j in range(1,repetition): # repeating frames

Nickduino, 2018-09-30:
You should probably indent the four _wf.append()_

But if you didn't get that, the learning curve to understand how the script works is goind to be pretty steep.

closed 2018-10-27 by Nickduino