Nickduino/Pi-Somfy #13: socket.error: [Errno 104] Connection reset by peer

MichaelB2018, 2018-10-27:
Trying to use this library. The code seems to work fine is I call 

def envoi_commande(telco, bouton, repetition):

with a repetition of **1**. But as soon as I put repetition 2, I get the following error:

```
Traceback (most recent call last):
  File "./somfy.py", line 141, in <module>
    envoi_commande("chambre", buttonStop, 2)
  File "./somfy.py", line 132, in sendCommand
    pi.wave_add_generic(wf)
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 2120, in wave_add_generic
    self.sl, _PI_CMD_WVAG, 0, 0, len(pulses)*12, extents))
  File "/usr/lib/python2.7/dist-packages/pigpio.py", line 1007, in _pigpio_command_ext
    dummy, res = struct.unpack('12sI', sl.s.recv(_SOCK_CMD_LEN))
**socket.error: [Errno 104] Connection reset by peer**
```

Any ideas?

Nickduino, 2018-10-27:
Hum, not much ideas I'm afraid.

Are you sure the pigpiod daemon is running? Have you checked the solved issues? https://github.com/Nickduino/Pi-Somfy/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aclosed

MichaelB2018, 2018-10-27:
Thanks. appreciate the quick response. I did a bit of digging and my pigpiod is in v64, which seems correct. However I got another thought. When I download the somfy.py from github and run it, I get an error message:

`  File "./somfy.py", line 115
    for j in range(1,repetition): # repeating frames
    ^
IndentationError: unexpected indent
`

So I added indentation to line 113, making essentially a nested for loop in line 115 (within line 105). But I'm not thinking whether I should remove indentation from line 115, bringing it on the same level as line 105, and therefore not nesting the 2 for loops.

What do you think is correct?

turuwhiska, 2018-10-27:
is only a problem of tab spaces

turuwhiska, 2018-10-27:
i have the file complete can i upload the file?

MichaelB2018, 2018-10-28:
Yes I did download it again. You see the same problem here with line 115 (following 113): 

https://github.com/Nickduino/Pi-Somfy/blob/master/somfy.py

turuwhiska, 2018-10-28:
for j in range(1,repetition): # repeating frames
         for i in range(7): # hardware synchronization
                wf.append(pigpio.pulse(1<<TXGPIO, 0, 2560))
                wf.append(pigpio.pulse(0, 1<<TXGPIO, 2560))
         wf.append(pigpio.pulse(1<<TXGPIO, 0, 4550)) # software synchronization
         wf.append(pigpio.pulse(0, 1<<TXGPIO,  640))

   for i in range (0, 56): # manchester enconding of payload data
      if ((frame[i/8] >> (7 - (i%8))) & 1):
          wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))
          wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
      else:
          wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
          wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))

          wf.append(pigpio.pulse(0, 1<<TXGPIO, 30415)) # interframe gap

   pi.wave_add_generic(wf)

Original file say:
      for j in range(1,repetition): # repeating frames
            for i in range(7): # hardware synchronization
                  wf.append(pigpio.pulse(1<<TXGPIO, 0, 2560))
                  wf.append(pigpio.pulse(0, 1<<TXGPIO, 2560))
            wf.append(pigpio.pulse(1<<TXGPIO, 0, 4550)) # software synchronization
            wf.append(pigpio.pulse(0, 1<<TXGPIO,  640))

            for i in range (0, 56): # manchester enconding of payload data
                  if ((frame[i/8] >> (7 - (i%8))) & 1):
                        wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))
                        wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
                  else:
                        wf.append(pigpio.pulse(1<<TXGPIO, 0, 640))
                        wf.append(pigpio.pulse(0, 1<<TXGPIO, 640))

wf.append(pigpio.pulse(0, 1<<TXGPIO, 30415)) # interframe gap
you need re-order this lines
and remember start pigpiod

turuwhiska, 2018-10-28:
The file is fixed, but for some reason the old file
you just have to modify the spaces of these lines and it works well
and do not forget to start pigpiod

Nickduino, 2018-10-28:
> The file is fixed, but for some reason git down the old file

That's a weird behavior. Anyway, yes the _wf.append()_ around line 126 have to be inside the _if:_ and the _else:_

closed 2018-10-28 by Nickduino