Nickduino/Pi-Somfy #16: About changing the saw oscillator

GatienC, 2019-01-03:
Hi !

I'm pulling another issue because I don't know how to change the saw oscillator and i cant' find anything on google. Do i have to desolder the actual oscillator and solder the new one or is there something else I don't get ? :)

Anyway thanks for everything !

lfliess, 2019-01-03:
You got it right, got to desolder the old one first

Nickduino, 2019-01-03:
Yes, it's that "simple"

Be prepared to spend a good amount of time on the desoldering job if you're not used to it

closed 2019-01-03 by Nickduino

GatienC, 2019-01-05:
Thanks ! Any advices about desoldering ? :)

Nickduino, 2019-01-05:
Heat one pad and try prying gently on the oscillator (with a toothpick, for instance). If it moves slightly, repeat with the second leg, then the third, then go back to the first...
Once you can cut its leads, do that and remove the remaining leads.

Or put a big blob of solder and heat all three pads and legs at once.

Or use the classic method, one pad at a time with desoldering braid.

Or whatever suits you, really

GatienC, 2019-01-06:
Thanks ! :) I'll try it next weekend

kjellhult, 2020-08-30:
When you re-solder the new emitter to the board, keep in mind that one pin is grounded to the metal case and the others have *very tiny* insulators around their pins.  Be sure to clean up your excess solder very well, and maybe put a tiny spacer between the case and the board to prevent any shorting to the other pins.