Nickduino/Pi-Somfy #20: Sunset-sunrise as argument

Nickduino, 2019-01-30:
The way I see it, the day starts in the morning and end in the evening, I'd prefer it if the first argument passed was to open he blinds (accepting a negative value if you want to open them before sunrise) the the second to close them.

[image]

Moreover, we may want to use the astronomical data at night but not in the morning (open at 7 every day, close right after sunset) so I think we should seperate that in three arguments:
-    ``-t      time``
-    ``-m    sunrise``
-    ``-n    sunset``

lfliess, 2019-04-22:
I'm working on adding support for Somfy Awnings.
for those, we need to do the opposite - you want to lower them (open) some time after sunrise, and rise them (close) some time before/after sunset.
So if anyone has time, making these changes would be great.

Nickduino, 2019-05-05:
We should definitely do that.
The temporary solution is to lower/rise them at fixed hours for the moment.

MichaelB2018, 2019-05-22:
Not sure I understand the ask here. Using the web interface it should be quite simple program what you suggest. The command line interface was never really meant for complex actions, but rather just as a debugging tool. But maybe I miss something. Have you tried doing the above using the web interface and add a "Scheduled Operation"?

lfliess, 2019-05-28:
You're right, didn't realize webUI works as expected.
CLI should have the same ability, but it's not a must

closed 2020-04-18 by MichaelB2018