Nickduino/Pi-Somfy #21: Improving the GUI

Nickduino, 2019-01-30:
I've made a 3D render of a streamlined remote:
[image]

The objective is to replace this:
[image]

Any one has some knowledge of web designing and wants to help?
Possibly responsive design as well but not necessarily

c6opuc, 2019-04-18:
First of all thank you guys for the great project and the level of maturity of the code and documentation! I managed to build the system that controls 7 blinds in just two days. The longest time took soldering for me and I could only overcome it by involving my friend :-) Otherwise it was extremely easy process and all worked from the first attempt! 
Is there any chance to revert the interface of operating the remotes to the old one? The new one is really not ready. There is overlapping of remote in Chrome on PC (see screenshot), and it is completely unusable on Android smartphone.

[image]

Nickduino, 2019-04-19:
Do you feel able to work on it?

If you want to revert back to the old way, have you tried the V2 ?
https://github.com/Nickduino/Pi-Somfy/releases

c6opuc, 2019-04-19:
If I revert to V2 I will loose all other fixes regarding alexa, etc, right? 
I am not a web-dev and already had a look at the code. It is too complicated for me :-)

Nickduino, 2019-04-19:
Well, Alexa should work with the last release.

For the rest, I know it's frustrating

PR #33 (2019-05-22): Web gui improvements
Proposal to fix #21 :
- No more overlapping when multiple rows
- Buttons now work with touchscreens
- More compact and "cleaner" responsive layout (although I'm far from an expert)
closed 2019-05-22 by MichaelB2018