Nickduino/Pi-Somfy #25: Rolling code glitch

Nickduino, 2019-02-02:
I don't know what happened:

[image]

The error:

[image]

I deleted line 108 and added 4 to the rolling code of line 98 for good measure. All seems well (the shutter works)

Nickduino, 2019-02-02:
Damn, happened again :-/

MichaelB2018, 2019-02-02:
Not sure I understand the problem,  so it created a duplicate with lower case?  Ie somewhere there is a lowercase/ uppercase conversion issue?

Nickduino, 2019-02-02:
1.  Yes
2.  And since there are two times the same address, it throws an error code.

MichaelB2018, 2019-02-03:
I checked in a fix that makes sure that shutter IDs are always upper case. 

But I'm not sure that's the root cause. Since your log file seems to indicate that there are issues with all the other config values as well. TXGPIO, etc. So I'm more concerned that the issue could be related unicode situation or special characters that got added to the config file and prevent the parser to understand the file. Did you copy the config file from Windows? Or anything else that could have caused this. Maybe best if you can send me the config file off-line?

MichaelB2018, 2019-02-03:
ok, confirmed the problem you saw was indeed only linked to the duplicate entry, which should be avoided with the uppercase conversion  of the shutter id introduced yesterday.

ssanden, 2020-05-25:
this issue seems not to be resolved.
i have 10 shutters. the last one is added a second time with in lowercase:
[image]

Nickduino, 2020-05-25:
Shit! I hasn't happened to me in maybe a year.

Short term, delete the last line, increment the new last line and restart.

ssanden, 2020-05-25:
this has happened to me twice now. i needed to resetup the pi, after i broke the sd card..
the first time i just removed the line with the uppercase address, and it worked fine from then on.

i have a feeling that the last line may be added later (maybe on a specific action like "down" or something)..

thanks for the quick reply!

Nickduino, 2020-05-25:
> i have a feeling that the last line may be added later (maybe on a specific action like "down" or something)..

Yes (i.e. it doesn't happen on setup) but that's definitely a bug

MichaelB2018, 2020-05-31:
@ssanden, can you give me a bit more information when it happened. Did you add a new shutter? Did you edit it?  Or did it happen during normal operation without your manual input? The more you can share, the better the chance I can reproduce and fix it.

Nickduino, 2020-05-31:
Every time it happened to me, it was during normal operation. 

Not sure if it was during manual operation or programmed operation but I lean towards programmed

Secarius, 2020-07-06:
every time i reload the config i get this error too:
```
sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e -m
Error in MyConfig:ReadValue: 0x279621: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279622: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279623: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279624: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279625: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279626: invalid literal for int() with base 10: 'None' : myconfig.py:160
```

ssanden, 2020-07-06:
sorry, i was somewhat buzy..

@MichaelB2018 : for me this happend shortly after i added a new shutter. then, while the first few actions (up/down/stop) this came up..

Nickduino, 2020-07-09:
> every time i reload the config i get this error too:
>
> ```
> sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e -m
> Error in MyConfig:ReadValue: 0x279621: [...]

Check what's on line 160 in myconfig.py

ssanden, 2020-07-09:
the same error (...myconfig.py:160) comes up on my system..

i already filed an issue, [https://github.com/Nickduino/Pi-Somfy/issues/68](https://github.com/Nickduino/Pi-Somfy/issues/68)
which was closed and directed here.

this is the content of myconfig.py at line 160:

```
   144	    #---------------------MyConfig::ReadValue-----------------------------------
   145	    def ReadValue(self, Entry, return_type = str, default = None, section = None, NoLog = False):
   146	
   147	        try:
   148	
   149	            if section != None:
   150	                self.SetSection(section)
   151	
   152	            if self.config.has_option(self.Section, Entry):
   153	                if return_type == str:
   154	                    return self.config.get(self.Section, Entry)
   155	                elif return_type == bool:
   156	                    return self.config.getboolean(self.Section, Entry)
   157	                elif return_type == float:
   158	                    return self.config.getfloat(self.Section, Entry)
   159	                elif return_type == int:
   160	                    return self.config.getint(self.Section, Entry)
   161	                else:
   162	                    self.LogErrorLine("Error in MyConfig:ReadValue: invalid type:" + str(return_type))
   163	                    return default
   164	            else:
   165	                return default
   166	        except Exception as e1:
   167	            if not NoLog:
   168	                self.LogErrorLine("Error in MyConfig:ReadValue: " + Entry + ": " + str(e1))
   169	            return default
```

Secarius, 2020-07-09:
`return self.config.getint(self.Section, Entry)
`

> Check what's on line 160 in myconfig.py

firewiremb, 2020-07-10:
I had the same Problem today. It happend during normal Operation.

```
pi@raspberrypi:~ $ sudo systemctl status shutters.service
● shutters.service - Pi Somfy Shutter Service
   Loaded: loaded (/etc/systemd/system/shutters.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2020-07-10 12:50:52 CEST; 2s ago
  Process: 5722 ExecStart=/usr/bin/sudo /usr/bin/python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a (code=killed, signal=TERM)
 Main PID: 5722 (code=killed, signal=TERM)

Jul 10 12:50:51 raspberrypi sudo[5722]: Error in MyConfig:ReadValue: TXGPIO: int() argument must be a string, a bytes-like object or a number, not 'list' : myconfig.py:160
Jul 10 12:50:51 raspberrypi sudo[5722]: Error in MyConfig:ReadValue: MQTT_Port: int() argument must be a string, a bytes-like object or a number, not 'list' : myconfig.py:160
Jul 10 12:50:51 raspberrypi sudo[5722]: Error in MyConfig:ReadValue: EnableDiscovery: 'list' object has no attribute 'lower' : myconfig.py:156
Jul 10 12:50:51 raspberrypi sudo[5722]: Missing config file or config file entries in Section Shutters for key 0x279621: 'list' object has no attribute 'split' : myconfig.py:78
Jul 10 12:50:51 raspberrypi sudo[5722]: Failure to load configuration parameters
Jul 10 12:50:52 raspberrypi systemd[1]: Stopping Pi Somfy Shutter Service...
Jul 10 12:50:52 raspberrypi sudo[5722]: pam_unix(sudo:session): session closed for user root
Jul 10 12:50:52 raspberrypi systemd[1]: shutters.service: Main process exited, code=killed, status=15/TERM
Jul 10 12:50:52 raspberrypi systemd[1]: shutters.service: Succeeded.
Jul 10 12:50:52 raspberrypi systemd[1]: Stopped Pi Somfy Shutter Service.
```
operateShutters.conf:

```
[ShutterRollingCodes]
0x279621 = 16
0x279622 = 20
0x279623 = 27
0x279624 = 19
0x279625 = 16
0x279626 = 21
0x279627 = 28
0x279628 = 25
0x279629 = 15
0x27962A = 8
0x27962a = 19
´´´

Nickduino, 2020-07-10:
Delete the last line and change the 8 to 20.

firewiremb, 2020-07-10:
Everything worked fine again after correcting the values as Nickuino mentioned. But now it already happend again.
Is there something I can do to support debugging?
```
[Shutters]
0x279621 = Bad,True,16
0x279622 = 9-Laura,True,20
0x279623 = 1-Wohnen,True,28
0x279624 = 2-Wohnen,True,22
0x279625 = 3-Essen,True,23
0x279626 = 4-Essen,True,21
0x279627 = 5-Essen,True,23
0x279628 = 6-Küche,True,15
0x279629 = 7-Küche,True,28
0x27962A = 8-Küche,True,18

# Indicates the rolling code used by the shutter, based on the address 
# provided in the section [Shutters]. Note that this changes every time 
# the remote is used and therefore the config file will update frequently
#
[ShutterRollingCodes]
0x279621 = 18
0x279622 = 22
0x279623 = 27
0x279624 = 21
0x279625 = 16
0x279626 = 23
0x279627 = 30
0x279628 = 25
0x279629 = 20
0x27962A = 26
0x27962a = 27
´´´

ssanden, 2020-07-10:
i'd suggest to remove the string that ends with the uppercase letter and keep the one with the lowercase end.
that worked pretty well for me so far.

Secarius, 2020-07-11:
I still get the error:
```
sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e -m
Error in MyConfig:ReadValue: 0x279621: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279622: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279623: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279624: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279625: invalid literal for int() with base 10: 'None' : myconfig.py:160
Error in MyConfig:ReadValue: 0x279626: invalid literal for int() with base 10: 'None' : myconfig.py:160
```
operateShutters.conf:
```
[Shutters]
0x279621 = Büro,True,15
0x279622 = Wohnzimmer,True,5
0x279623 = Balkon,True,1
0x279624 = Küche,True,1
0x279625 = Kinderzimmer,True,1
0x279626 = Schlafzimmer,True,17

# Indicates the rolling code used by the shutter, based on the address
# provided in the section [Shutters]. Note that this changes every time
# the remote is used and therefore the config file will update frequently
#
[ShutterRollingCodes]
0x279621 = 60
0x279622 = 53
0x279623 = 37
0x279624 = 44
0x279625 = 38
0x279626 = 39

# Indicates in % the intermediate position stored in the shutter itself,
# based on the address provided in the section [Shutters].
# Bellow values are assumed to match with shutters internal presets.
# They might be overwritten/cleared by a 5 seconds press on the stop button.
#
[ShutterIntermediatePositions]
0x279621 = None
0x279622 = None
0x279623 = None
0x279624 = None
0x279625 = None
0x279626 = None
```

firewiremb, 2020-08-09:
@ssanden 
Thank you for your hint. This works for me too.

@Secarius 
The reason for the error is the initial declaration of [ShutterIntermediatePositions] values as _None_ in the Config File.

The Object _config.getint(self.Section, Entry)_ in Line 160 can't handle the Type None.

To get around this problem quickly, you can set the values to e.g. -1 or 101, as you obviously don't use shutter intermediate positions.

Secarius, 2020-08-10:
@firewiremb Thank you, but what is the ShutterIntermediatePositions?
I don't understand, what this value is for

firewiremb, 2020-08-10:
@Secarius 
I don't know if all Somfy systems have this feature. In the installation in my house, motors are used that can save an intermediate position, so to speak, "in the shutter motor".

If you press the stop button in Pi-Somfy when the shutter is stationary, the shutter moves to this saved position. However, Pi-Somfy does not know the position stored in the motor. So that Pi-Sofmy then knows the position for further movements of the shutter, this must be stored manually in the configuration file.

Secarius, 2020-08-10:
@firewiremb yes, when i press the stop button on my wall switch the motor moves the shutter to +- 50 percentage.
So what number should i set? (50) for 50%?

MarkGodwin, 2021-12-11:
Just ran into this now I have more than 10 remotes.
The easiest way to fix it is to change all the ids in the config file to lower case. It's only the initial setup actions in the web UI that write the IDs as uppercase. All of the later writes & updates (via Mqtt and Web UI) come through with lower-case Ids. So even an edit in the web UI will create a duplicate entry for the shutter as lowercase.
I do have a better fix for this in my repo, but my fork has diverged quite a bit, so I would need to do a bit of work to extract the relevant changes for a PR.

Nickduino, 2022-01-02:
> I do have a better fix for this in my repo, but my fork has diverged quite a bit, so I would need to do a bit of work to extract the relevant changes for a PR.

Is that on your radar or not at all?

MarkGodwin, 2022-01-03:
So my change for this is mixed in with changes to the MQTT API that worked better with Home Assistant (for me, anyway), and a feature to keep the status of the blinds in slightly better state if you have multiple remotes controlling sets of blinds.
I kind of doubt you want those other two changes as the MQTT one will break compatibility with anyone that isn't using Home Assistant, and the shutter grouping is a bit of a niche requirement where I really wanted all the blinds to start moving at exactly the same time, but I also wanted individual control of each blind. So I have one remote that controls a set of the blinds, but the system needs to know that the individual blinds are also affected when the shared blind is activated. This worked better for me that HA's grouping feature.
I will try to find some time this weekend to create a PR for the case sensitivity problem.
I can do PRs for the other features if you want them.

Nickduino, 2022-01-03:
I'd like to know what the others think but I think being able to track a blind state when moved from multiple remotes is useful

MichaelB2018, 2022-01-03:
Keen to know what you changed for MQTT? Does it break MQTT discovery?

ssanden, 2022-01-03:
personally, i would prefer to have the basic commands (up, down, stop, program and the hardware/gpio-stuff) in a seperate "core package". that package should then be usable like an api.. or maybe be importable or whatsoever..

all the higher functions (partial rise/lower, tracking, webinterface, mqtt, alexa and so on) would then also live in a seperate entity and just make use of that core package.

i dont know if this is practicable.. and it may also be a bit selfish since it just fits my requirements the most :)
but anyway.. imho the outcome could be expected to be cleaner, easier to maintain and less complex and would scale better as the project grows..

of course this is a lot of work, that means to change a great deal.
just throwing in that random thoughts, maybe useful for some future refactoring. feel free to ignore this if you disagree :)

greetings

rbswift, 2022-01-03:
Agreed. The position tracking code should ideally be seperate from the signalling code. This would need a refactor. Also as mentioned, the presumption of one channe=one motor is not always accurate and in fact this can be many-to-many although that does add unnecessary complexity for most users

Nickduino, 2022-01-05:
> personally, i would prefer to have the basic commands (up, down, stop, program and the hardware/gpio-stuff) in a seperate "core package"

I don't know how basic you like things but my V1 release before @MichaelB2018 came on board was extremely basic :
https://github.com/Nickduino/Pi-Somfy/releases/tag/v1.0

MarkGodwin, 2022-04-24:
> Keen to know what you changed for MQTT? Does it break MQTT discovery?

It improves the HA MQTT discovery, by changing the blind into the sort that has an an opening, closing and stopped state, as well as a position value which can be requested via a different endpoint. That makes HA work a bit better, as it can enable the Open, Close & Stop buttons at the right times, track (sort of) the blind position, and request specific positions. But it changes the MQTT API completely so anyone not using HA discovery would need to update their commands. This is why I didn't consider a PR. You'd probably want to have a config switch to retain the original format.

Most of the changes are in this [commit](https://github.com/MarkGodwin/Pi-Somfy/commit/e8fb0b887651fafbdfb7bbd2fd99c92749437188).

gbsallery, 2024-02-05:
Just as another datapoint, this issue is still occurring for my setup as well. It means periodically logging into the Pi, deleting the spurious value, and re-starting the service. I usually realise that I need to do this when the blinds fail to open in the morning :-/

Sometimes it takes weeks between glitches, sometimes only a day or two, but a fix would be welcome (or even just some pointers as to where to start debugging this; I'm not really familiar with Python, but can usually find my way around a codebase eventually).