Nickduino/Pi-Somfy #28: Latest master (5b80b56) is broken

tholu, 2019-02-07:
```
pi@raspberrypi ~/Pi-Somfy $ sudo python /home/pi/Pi-Somfy/operateShutters.py -a

This program requires the modules located from the same github repository that are not present.

Error: Non-ASCII character '\xc2' in file /home/pi/Pi-Somfy/myalexa.py on line 74, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details (myalexa.py, line 74)
```

The offending line is:
```
            self.LogInfo ("Remote address in dec: " + str(int(shutterId,16)) + ", WeMo port will be n°" + str(portId))
```

Should be easily fixable.

Nickduino, 2019-02-08:
Probably missing this on one of the scripts:
`# -*- coding: utf-8 -*-`

I couldn't see it as my system is UTF-8 by default

Nickduino, 2019-02-08:
Edited, please try again

tholu, 2019-02-08:
My system was UTF8 as default as well:

```
pi@raspberrypi ~ $ locale
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```

It now works, thanks!

closed 2019-02-08 by tholu