Nickduino/Pi-Somfy #29: (NOT and issue) Not sure how to contact you.

mjessup, 2019-03-11:
I was wondering how/if it is possible to add IFTTT functionality to this. If possible I'm looking to get some small door sensors that will go on the blinds and will integrate with stringify.

Thanks.

Nickduino, 2019-03-11:
IFTTT would be a useful addition to the project!

1. If you're a developer and want to it, we'd be happy to integrate your implementation through a pull request.
2. If you don't code, you'll have to find a workaround. Maybe (I've never used it, so...) you can use Stringify to launch an Echo routine (with will control the fauxmo device)? Or launch a Python script through the Maker Channel on IFTTT?

mjessup, 2019-03-11:
I'm by no needs a developer but I know the very basics. I did some research and it might be possible to run a terminal command on the raspberry pi to open/close a certain blind. I found this project called Triggercmd. I did bring up in my original post how I plan to use cheap door sensors. This is because at least when using the stock somfy remote it doesn't work 100% of the time, I plan to set up something using stringify to make sure the correct commands are sent to the blinds.

Nickduino, 2019-03-11:
Let's start from the beginning, then.

What are you trying to achieve? Do you really need IFTTT or can you automate the opening/closing of your shutters differently? If it's for a manual control, why not use the GUI Michael coded?

"Not 100% of the time"...is it 99.99%, though? Sending the command twice would mean a success rate of 99.999999%

mjessup, 2019-03-11:
My whole house is controlled with IFTTT and stringify so IFTTT is a must. I didn't think about sending the command twice thanks for recommending that. I feel stupid for not thinking of sending two commands. I'm just concerned how good the IFTTT integration would be simply using Triggercmd like how would I set up that to send the right toggle (up/down).

closed 2019-03-12 by Nickduino

lfliess, 2019-04-22:
Actually, IFTTT would be great! say you want to act according to weather (adding this code now), with IFTTT we don't need to program new logic- IFTTT already allows weather based desicions.

Nickduino, 2019-05-05:
@lfliess Can you code it?