Nickduino/Pi-Somfy #31: Alexa integration: only up/down is possible

c6opuc, 2019-04-19:
Hi guys,

I was wondering if you plan to enhance Alexa integration by adding the possibility to stop the blinds? At the moment I can only make them either completely opened or closed which is rather limited :-( 
Thank a lot for your work! Great project!

Nickduino, 2019-04-19:
The Alexa integration is done through a WeMo outlet emulation. The real outlets having only two modes (on or off, they're not quantum outlets), that's why it was implemented this way.

It would, off course, be possible to create another fauxMo device just for the sake of stopping the blinds/shutters but that's more work and I'm not sure it would be all that useful because of the latency inherent to the voice recognition.

c6opuc, 2019-04-19:
Hi Nickduino,

I agree with you with regard to the voice recognition and perhaps it is not usable when commanding with voice, however you can still use the Alexa app to do fine control with buttons. One can argue that same can be achieved differently, for instance with openHAB, but the command line interface is also very slow. I find that webinterface in conjunction with alexa integration is optimal combination.

Nickduino, 2019-04-19:
The Alexa GUI should still be a bit slower than using Pi-Somfy directly. And adding a device to stop the shutters would be much worse: send a down command through the Alexa app, close this device control, open the "stop" faux device, send the stop command.

Besides, do you find the Alexa app usable at the moment? If you open the shutters with a remote or a rule, you can't close it with the Alexa app.

closed 2019-07-07 by MichaelB2018