Nickduino/Pi-Somfy #32: weather integration for awnings

lfliess, 2019-04-22:
I'm starting work on a weather module - integrates with Yahoo weather, and allowing user to register for closing the awning in case of rain/strong wind (wind speed configurable).
I will add this to work together with the CLI shcedule, but would need help to integrate it into the GUI.

Nickduino, 2019-05-05:
We already have some weather implementation in the pipeline with Michael. It's likely it will do what you expect but the development is pretty slow, sorry for that.

lfliess, 2019-05-12:
in that case, I'd like to offer my help since I already have some code done.
@MichaelB2018 - do you want to sync on this?

MichaelB2018, 2019-05-12:
sure - have not had much time to look into this recently, but feel free to merge it.

lfliess, 2019-05-14:
Is there any partial code you want me to look at? or just go ahead with what I started and we can see where we want to take it from there?

MichaelB2018, 2019-05-22:
no partial code, I'm afraid. The only thing 2 things I was thinking off are:
1.) use https://github.com/csparpa/pyowm
2.) add a timeType of "weather" (currently it has clock & astro) in the scheduler

Also a small hint.. watch out of oscillating values in OpenWeatherMap... eg. the below is a graph of wind speeds... If I had set a trigger to for the awning to lower/raise at a wind speed of 6 m/s, my awning would have been non-stop busy today:
[image]
So maybe add some threshold or similar.

Nickduino, 2019-05-22:
In that case, I think the easiest thing is to use hysteresis. You' close the awning if the wind exceeds 6 m/s and open it if it drops below 2 (or don't open it at all)

MichaelB2018, 2019-05-23:
which you can do easily with 2 different "schedules", so you don't even need to cater for hysteresis in the code.

Nickduino, 2019-05-23:
Absolutely

lfliess, 2019-05-26:
1. I started with yahoo weather, but I can use OWM just as well
2. not really a scheduler as that isn't safe - strong winds can start suddenly. I want to act according to live weather.
I can just poll the weather every few minutes, and if the wind is high, or if it rains, just close the awning.
3.I have the wind part done, but having some trouble with rain as both Yahoo and OWM only tell you if it's raining using strings ("sunny", "partly cloudy", "raining", etc...). but there are so many different options for raining that for this I might go with a scheduler - decide on a daily basis to open or not (if it's not sunny or even raining - no need to open awning at all)

MichaelB2018, 2019-05-27:
1. & 2.) perfect, no worries
3.) Maybe even languages might be a concern as well for some folks. But I like the idea of a scheduler for certain weather and immediate events. We were talking once about adding conditions to existing time or astro related events.  Maybe add something in the Event class.

lfliess, 2019-10-28:
I gave up on this.
will use the new MQTT support to have Home Assistant do this. it already has good weather integration that's being maintained.
I actually think this is a better solution to everyone. you can even run hass.io on the same RPi that you use this for. then you don't need MQTT, you can just call the command line commands....

an interesting project might be to implement Pi-Somfy as a HA plug-in

closed 2019-10-28 by lfliess

MichaelB2018, 2019-10-28:
Make sure you use MQTT discovery for Home Assistant (https://www.home-assistant.io/docs/mqtt/discovery/) which makes it easier. Probably need to do a bit more documentation when I get to it. But let me know if you get it to work. Keen to get some more testers in case of any issues

lfliess, 2019-10-29:
I tried it yesterday with my newly installed hass.io with Mosquitto
I installed it in a docker on the same rpi

but I'm getting trouble with discovery.
from the logs so far it seems your code does the job and sends a message that looks like a legitimate discovery message, but hass.io doesn't discover a new device.

I also tried manually adding the topics but I think I now have the problem where the MQTT broker starts up too late and hass.io gives up on it on startup.

will try to resolve my issues and tell you if it works for me

lfliess, 2019-10-29:
I didn't configure my broker correctly.
now it works. the cover was discovered and I'm able to control it correctly through HA

great job!
I can now use the openweathermap hass.io integration to create any automation I'd like.
this is just what I needed

MichaelB2018, 2019-10-29:
glad t worked. Given that it;s all fresh in your mind, do you want to help with documenting how to set it up? (as you can see, documentation is never my favorite part).

lfliess, 2019-10-30:
Sure
I did just what the documentation said and it worked (-m flag)
The trouble I had was setting up Hass.io with mqtt
I can put in paragraph about it...

Nickduino, 2019-10-30:
> The trouble I had was setting up Hass.io with mqtt I can put in paragraph about it...

Definitely.