Nickduino/Pi-Somfy #35: Home assistant/openhab/ Mqtt support

engrfash, 2019-06-29:
Is there no MQTT/OPENHAB/HOME ASSISTANT support?

MichaelB2018, 2019-06-30:
Not at the moment.  Would love for somebody to volunteer to add this.

MichaelB2018, 2019-07-06:
@engrfash. Was thinking a bit more about this. The Python code to publish to MQTT would be quite simple. But there are 2 difficulties:
1.) Your status would not show updates from the original remote control - ie when a user uses the factory supplied remote to lower or higher it. I did think once on what it would take to use the receiver to listen to those other remotes. But I did not get very far. Would love to hear it if anybody was successful
2.) The GUI work to enhance the "Setting page" to allow for manual config updates is a bit of work. Not sure anybody would be keen to help with this?

arnauos, 2019-08-08:
I have integrated this to Apple Homekit as simple push buttons for UP/DOWN/STOP for each blind/awning... not perfect but it works.

I'm sending http requests through homebridge so if MQTT/OPENHAB/HOME ASSISTANT can also send those, this could also be a possible solution.

suhajda3, 2019-11-16:
Could you specify how to use the HTTP requests. I am planning to use it with [homebridge-blinds](https://github.com/zwerch/homebridge-blinds). Thank you!

rbswift, 2019-11-16:
see my homebridge-blinds config here.... https://github.com/Nickduino/Pi-Somfy/issues/36

..but I've since switched to homebridge-mqttthing because it gives more control

MichaelB2018, 2020-01-07:
closed, as this is now implemented

closed 2020-01-07 by MichaelB2018