Nickduino/Pi-Somfy #36: HomeKit support

jonchargreaves, 2019-08-01:
Hi all,

This is an amazing app just wondered is there consideration for HomeKit development?

MichaelB2018, 2019-08-02:
Not in progress at the moment, But an MQTT interface has been in discussion before and definitely looks feasible. More volunteer would help to speed up the process

jonchargreaves, 2019-08-03:
Happy to assist with any testing but programming is beyond my skills.

rbswift, 2019-08-04:
You can use homebridge with the BlindsHTTP plugin, configured as follows...
```
  "accessory": "BlindsHTTP",
  "name": "Lounge Blind",
  "up_url": "http://127.0.0.1/cmd/up?shutter=0xXXXXXX",
  "down_url": "http://127.0.0.1/cmd/down?shutter=0xXXXXXX",
  "stop_url": "http://127.0.0.1/cmd/stop?shutter=0xXXXXXX",
  "motion_time": "20000",
  "http_method": "POST",
  "trigger_stop_at_boundaries": false
```

closed 2019-08-04 by MichaelB2018