Nickduino/Pi-Somfy #38: pigpiod constant cpu usage

rbswift, 2019-08-16:
Hi, is there any way to reduce the constant cpu usage by pigpiod? This process uses about 7-10% cpu on my pi constantly even when pi-somfy is not being used. Cheers

rbswift, 2019-09-17:
I found this unnecessary CPU usage seems to be eliminated by running pigpiod with the -m switch. I'll test some more and then try a PR

Nickduino, 2019-09-17:
What does it do? Stops listening?

Congratulations for finding your fix, by the way

rbswift, 2019-09-17:
Cheers Nick. Thanks for your work on the project!

Seems like, by default, pigpiod samples/polls all 32 GPIO as inputs every 5us and generates alerts. Since Pi-Somfy is not using inputs, only outputs, I think sampling can safely be disabled with the -m switch. This reduces CPU use from around 7-10% to essentially insignificant.

I might have a go at adding mqtt with the paho library as a PR too.

Nickduino, 2019-09-18:
I personally used pigpio first to **receive** messages from another home automation project (my mailbox), so I can't disable it. Maybe I should check the documentation and listen to only one pin, though

closed 2019-09-18 by Nickduino