Nickduino/Pi-Somfy #39: Web GUI doesn’t work on iOS

ToddGreenfield, 2019-09-16:
On my iPhone running safari web browser I navigate to the web url and am unable to click the manual operation tab, so I can’t see the up/stop/down buttons.  The only thing I see is +Add New for a new scheduled event. Clicking on settings or add/remove shutter doesn’t work either. Everything else works great on my laptop and with Alexa, just unable to use my iPhone browser. I also tried chrome on the iPhone with same results. Suggestions?

FraBoCH, 2019-09-16:
As I'm the last one to have modified the web GUI, I'll take a look at your issues and try to fix them.

FraBoCH, 2019-09-17:
I have a fix for that issue in a local branch. Simple HTML change.
@Nickduino can you give me permissions to push ? I would like to avoid creating a fork simply for that ;-)

Nickduino, 2019-09-17:
> As I'm the last one to have modified the web GUI, I'll take a look at your issues and try to fix them.

Thanks!

> I have a fix for that issue in a local branch. Simple HTML change.

Congrats!

> @Nickduino can you give me permissions to push ? I would like to avoid creating a fork simply for that ;-)

Done

closed 2019-09-17 by Nickduino