Nickduino/Pi-Somfy #42: Reading normal Somfy Remotes

matheuscanela, 2019-09-20:
Hi guys,

I haven't tested yet this project, but I was using a simple one which could open and close my Somfy blinds. So, your project seems to better, congratulations!

I know that the somfy motors won't provide a feedback to allow us to know the blinds position, like 30% down. We can only request down, up and stop, but no feedback.

Using a timer, we can predict where the blinds are as the raspberry pi will send down/up and the code will know how long it takes to roll over. Easy I guess.

However, if someone has used the traditional remote control, the raspberry will be out of sync and it is not going to work.

I am planning to develop an extension of your code to use the receiver 433Mhz to receive the original somfy remotes input. So, if someone uses the remote to put the blinds down/up, the raspberry pi will be able to receive this information and track the blinds position. Have you guys tried to use the receiver that comes together with the transmitter?

I would like to have this feature as I will put the blinds down based on the sunlight coming from the window to ensure that I won't damage the furniture and at the same time keep a bright room .

What so you guys think?

I will also put this to work with HA, so everything will be manageable and automatized.

Thanks guys

Nickduino, 2019-09-20:
My instant feedback below, I'll try to sum-up what you want to do with my first input in a following post.

> I haven't tested yet this project, but I was using a simple one which could open and close my Somfy blinds. So, your project seems to better, congratulations!

The other project was using a Raspberry Pi too? Do you have a link to it?

> Using a timer, we can predict where the blinds are as the raspberry pi will send down/up and the code will know how long it takes to roll over. Easy I guess.

Easy-ish. You'll need to teach the script how long it takes to go up/down (probably not the same speed going up or down) for every blind, which needs creating a method for that, ideally adding it to the GUI, allowing the GUI to re-register it if necessary. Is the speed consistent with the seasons and sunlight (exterior temp)?

> I am planning to develop an extension of your code to use the receiver 433Mhz to receive the original somfy remotes input. So, if someone uses the remote to put the blinds down/up, the raspberry pi [...]

I personally use it for something else (get a notification when the mailman drops mail or a package). The good news is that pigpio can listen too and you'll access this functionality with a Python too. The other good news is that I'm 99.9% sure it will receive the codes from the real Somfy remotes because these receivers are reasonably tolerant frequency-wise.
If I reckon correctly, I even used it myself (plugged to an Arduino) to check the frames I sent were identical to the ones of a real remote (before you ask: nope, that sketch is lost forever).

> I will also put this to work with HA, so everything will be manageable and automatized.

Excellent! We were contemplating the integration to a home automation solution but didn't know where to start as none of us use one...

Nickduino, 2019-09-20:
You say "easy" but I reckon your objective is somewhat ambitious. I would cut it into several different more-digest projects:

1. Track the state of every blind
2. Learn how to read the real remote
3. Know where the sun is: [PyEphem](https://pypi.org/project/ephem/) (that we already use to know sunrise/sunset) will help you track the sun: I assume knowing at what angle it will hit the window will help you decide how low you need the blind to be lowered.
3bis. You can't live in the shade all year round. Protecting your windows (and inside temp) from the sun is useful only when the UV index is pretty high. The best free solution I've found for that is UVindex, which works nice with Python (https://github.com/bachya/pyopenuv/blob/bd6dd998d4f0b2e0ec017b6d1bf673fc5bfb5531/example.py) and will allow you 50 requests a day. I'll add a simple example to the next post.
4. Include it to Home Automation.
5. Change the GUI to include all that (state tracking, address of the actual Somfy remotes, what blinds they control, how much is too much UV, what sunrays angle is alright for what blind at what season, what elements to make available to HA, etc)

I would suggest to start including openUV to the current project (3bis) and use it to create conditions such as "if max UV index today is above 7, lower this blind and stop after 9 seconds". In my opinion, it's the simplest thing to do and will give you 90% of the functionality you are looking after.

1: That's useful and the elegant way of doing things but you can also create rules that lower a blind when the sun is blasting in the room and raise it back up afterwards.
2. You may never need this. I've noticed I never use any of my remotes...except at the heart of summer when I can't bear the heat. If you make the blinds smarter in august, you may never need to tough an actual remote ever again...not a priority at all
3. That seems overkill but pretty cool
4. Now, maybe you should start with that and do all the secret magic in HA (sun tracking etc.)?
5. This is a shit ton of work. The simplest the project, the less work the GUI will require

Nickduino, 2019-09-20:
As promised, the openUV example:

`#!/usr/bin/python3`

`import asyncio`
`import pyopenuv`
`import sys`

`from aiohttp import ClientSession`
`from pyopenuv import Client`
`from pyopenuv.errors import OpenUvError`

`async def main(api, lat, lon, alt = 0):`
`    try: #Create the aiohttp session and run the example.`
`        async with ClientSession() as websession:`
`            client = pyopenuv.Client(`
`                api,`
`                lat,`
`                lon,`
`                websession,`
`                altitude=alt)`
`
      
`            cur = await client.uv_index()  # Get curent UV information`
`    except OpenUvError as err:`
`        print(err)`

`    max = cur.get("result").get("uv_max") # Get the maximum UV index for the day`
`    print("Maximum UV index today: " + str(max))`
`    return max`
      
`asyncio.get_event_loop().run_until_complete(main(sys.argv[1],sys.argv[2],sys.argv[3]))`
(Launch with _python3 openUV.py "[API key]" lat lon_)

You can also ask it the window of time during which SPF protection should be used:
`print(await client.uv_protection_window())`

MichaelB2018, 2019-09-21:
A few points:
1.) Partial lift and lower (based on timing) is already implemented: 
[image]
I think we never updated the main screen shot...

2.) HA integration. We were discussing this a few time and the current thought is for an MQTT based (paho) implementation, maybe using discovery. I have a similar application using this and I was thinking of integrating this to Pi-Somfy. But if you think of a direct integration (not MQTT based), that would be great too.

3.) Receiver... That was the main reason why (2) was never been implemented. Unless we can capture the manual commands from the other remotes, HA in my case will usually state the wrong state, so pretty useless. If you can manage to solve the receiver, that would be great!

matheuscanela, 2019-09-27:
Hey guys!
Thank you for all messages, I have been busy these days, but I will come back to implement this and let you guys know.

I have already cloned the repo :)

MichaelB2018, 2019-10-20:
Thinking of doing the mqtt integration sooner or later.  Have you started. If mot, I was planning to close the alexa integration and add a "-m" option. Any thoughts? don't want to duplicate your work if you are almost ready.

matheuscanela, 2019-10-23:
Hey mate, not yet, we had a baby who is consuming my time hahaha, but I will do

MichaelB2018, 2019-10-27:
congrats. no worries. I've done the mqtt integration and will test it the next week or so before publishing it.

MichaelB2018, 2019-10-27:
ended up checking it in already. Have a look when you find time. Also added some changes for Home Assistant's MQTT discovery, and made it look like one of the the newer ZWave "covers"  that can set the level to a certain percentage. To do so, I had to make some changes to the timing functionality. But kept it backward compatible

closed 2020-01-07 by MichaelB2018