Nickduino/Pi-Somfy #44: Conflict with official remote?

alxlaxv, 2019-10-29:
Hi,
I was using the new webinterface and planification for weeks and now it just stopped working.
I touched nothing but I recently used the official remote manual.
Is there any kind of conflict possible?

Should I remove and reprogram it now to have it back working?

Thanks

Nickduino, 2019-10-29:
SSH into the Pi, then open the config file:
[image]

Go check the rolling codes section:
[image]

At the bottom of the list, do you have duplicate remote address?
(That would be in lower case)

alxlaxv, 2019-10-29:
Hello,
In this section I have:
```
[ShutterRollingCodes]
0x279621 = 151
0x279622 = 7
```

Nickduino, 2019-10-29:
That seems alright

> I was using the new webinterface and planification for weeks and now it just stopped working.

Do you mean you can't access the web interface anymore or that a button press stopped working?

alxlaxv, 2019-10-30:
The recurrent task to open shaders in the morning just stopped to be executed.
So I have open the web interface and the buttons inside do nothing now.

Nickduino, 2019-10-30:
After a button press on the GUI, the number in [ShutterRollingCodes] is incremented properly?

Have you checked your wiring? Any chance the emitter or antenna is faulty?

alxlaxv, 2019-10-30:
Yes properly incremented. I will double check the wiring tomorrow as nothing changed there.

alxlaxv, 2019-10-31:
Wiring is nothing, untouched.

MichaelB2018, 2019-11-02:
If it properly increases the counter, it seems to me quite unlikely that the software broke. You say the wiring on the Pi is ok and the Shutter works fine with other remotes... then can you try to bring the Pi closer to the Shutter. I did see some fluctuations on how far it can reach.

alxlaxv, 2019-11-08:
So, I have removed the whole cloned folder and cloned it again.
I tried to relaunch as-is and it says paho-mqtt is missing.
I have reinstalled all required packages and now it still doesn't react...
I'm a bit lost.

Nickduino, 2019-11-08:
It's pretty hard to help you without more information.

I'm still inclined to think it could be a hardware problem. Wiring issue, cold solder joint, out-of-tune antenna, dead emitter, ...

closed 2020-01-07 by MichaelB2018