Nickduino/Pi-Somfy #45: sendCommand repetition for Li-ion battery motor support

rbswift, 2019-11-07:
Hi,

Great project!

Is there a reason the SendRepeat parameter in the conf file is not used by operateShutters.py? It looks like repetition is hardcoded at 2 (or 1 for programming).

I've found the only way to get Li-ion battery powered motors to reliably wake up and process a command is to repeat the command 5 times. The genuine somfy remote also seems a bit less responsive with battery motors so I guess this is just a 'feature' :)

Even better, would it be possible to add a repetition parameter for each motor? Wired RTS motors don't seem to need repetition as much as battery powered motors.

Cheers,
Richard

rbswift, 2019-11-07:
actually, here's my hack... it's a bit worse than I first thought lol....

<         if shutterId == "0x888888": #battery operated motor seems to be a bit deaf!
<            self.sendCommand(shutterId, self.buttonDown, 5)
<            time.sleep(1)
<            self.sendCommand(shutterId, self.buttonDown, 5)
<         else:
<            self.sendCommand(shutterId, self.buttonDown, 2)

Nickduino, 2019-12-02:
You're right, sendCommand is called with two repetitions 
`self.sendCommand(shutterId, self.buttonStop, 2)`

rbswift, 2020-01-31:
Would it be possible to parameterise this resend value per blind rather than have it hard coded?

MichaelB2018, 2020-04-18:
Finally got to do this and checked in a  fix. See SendRepeat in the configfile. You can define it there.

closed 2020-04-18 by MichaelB2018

rbswift, 2020-04-18:
Thanks for the change. I need to set this individually, "per blind" though because the battery powered Somfy blinds need several more repeats to wake them up whereas the regular mains powered blinds do not. Is that possible for a future change?

rbswift, 2020-04-18:
Actually I just tried this and even with 5 repeats the battery shutters still don't work. It seems they need the little 'sleep' in my hack comment above which is a bit crude. Any suggestions on how to implement this more efficiently? Perhaps a new flag in operateShutters.conf [Shutters] section to indicate which shutters need awakening?

MichaelB2018, 2020-04-19:
you can try to add a delay of 1 second between retries and let me know if it works. Just open operateShutters.py and add the following at line 292: 
time.sleep(1)

If it works, I can add it to the main branch.

rbswift, 2020-04-19:
Thanks Michael, That seems to work. I'm not sure how long it takes the battery blind to go into sleep mode so will test again later but it's looking good so far

MichaelB2018, 2020-04-19:
test it a few days and if it works we can add it to the project. Let me know. BTW would 1/2 second work too?

rbswift, 2020-04-19:
Thanks. 1/2 second seems to be working too. I will definitely test and find the best option. Cheers

rbswift, 2020-04-23:
I've been running this with time.sleep(0.25) for a few days not and it seems to work sell so perhaps that could be added to the next release? Many thanks.

MichaelB2018, 2020-04-25:
thanks for testing. I checked in an enhancement.