Nickduino/Pi-Somfy #46: What to do if alexa doesn't discover shutters?

kperets, 2019-11-24:
Hi, great project!
I tried to start the server multiple times with -a-e switches, but alexa doesn't seem to discover any shutter.
The webui works and operates well on my defined shutters. Any advice here?

Nickduino, 2019-11-24:
Have you tried turning her off and on again?

MichaelB2018, 2019-11-24:
can you check the log file to see if you have  any entries in there for "Alexa":

`grep Alexa /var/log/operateShutters.log`

kperets, 2019-11-24:
@Nickduino restarting Alexa caused a subset of the shutters list to be discovered(3/10)
@MichaelB2018 The log is showing these lines about 8 times:
`2019-11-24 21:33:17,312 : [DEBUG] (Alexa     ) HANDLE REQUEST: b'POST /upnp/control/basicevent1 HTTP/1.1\r\nSOAPACTION: "urn:Belkin:service:basicevent:1#GetBinaryState"\r\nContent-Type: text/xml; charset="utf-8"\r\nAccept: \r\nUser-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6437)\r\nHost: 192.168.1.110:54346\r\nConnection: Keep-Alive\r\nAccept-Encoding: gzip\r\nContent-Length: 299\r\n\r\n<?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>'
2019-11-24 21:33:17,314 : [DEBUG] (Alexa     ) SEND RESPONSE: POST /upnp/control/basicevent1 HTTP/1.1\r\nSOAPACTION: "urn:Belkin:service:basicevent:1#GetBinaryState"\r\nContent-Type: text/xml; charset="utf-8"\r\nAccept: \r\nUser-Agent: Dalvik/2.1.0 (Linux; U; Android 7.1.2; AEODN Build/NS6437)\r\nHost: 192.168.1.110:54346\r\nConnection: Keep-Alive\r\nAccept-Encoding: gzip\r\nContent-Length: 299\r\n\r\n<?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>`

Nickduino, 2019-11-24:
I found that Alexa has a hard time discovering many devices at once. Have you tried a new discovery since?

kperets, 2019-11-24:
I have, but it couldn't discover more. I'll try to comment out the devices and add them one by one with discovery, maybe it'll help. I'll update...

MichaelB2018, 2019-11-24:
The log files looks good. @Nickduino is right thought, I too have seen problems with discovery before. did you check the log file during the time when you run discovery?

rgtrex, 2019-12-26:
I'm facing the same problem(Alexa doesn't discover my shutters). Is there something else I can do?
I'm using Raspberry-PI 4. Alexa script is good for RP4 version? 
Thx U all

rgtrex, 2019-12-31:
New Update...

I was able to make progress on the subject.
Whenever I ask Alexa to _Discover Devices_ I get the following lines:

**2019-12-31 06:59:33,260 : [DEBUG] (Alexa     ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 10.100.102.10:54337\r\nAccept: */*\r\nContent-Type: application/json\r\n\r\n'
2019-12-31 06:59:33,260 : [INFO] (Alexa     ) Responding to setup.xml for Kitchen Shutter
2019-12-31 06:59:34,294 : [DEBUG] (Alexa     ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 10.100.102.10:54338\r\nAccept: */*\r\nContent-Type: application/json\r\n\r\n'
2019-12-31 06:59:34,295 : [INFO] (Alexa     ) Responding to setup.xml for Side Balcony Shutter**

Alexa seems to be able to contact Raspberry Pi but Raspberry Pi does not "answer" her. And of course Alexa doesn't find any device (shutter). Makes sense?

More info:
Raspberry Pi 4
Echo Dot 3rd

Maybe port issue? or python version issue?
Fauxmo.py- File not suitable for working with Echo Dot 3rd?

What can I do to solve this problem?

nstone33, 2020-01-07:
I have got blind control to work via manual control on the web interface but been unable to get Alexa to recognise the two blinds. After the first few failures with Alexa I removed one of them from the setup and tried to get Alexa to recognise one at a time, but that didn’t help. 

I am getting a repeated message in the command line window of “Trying not to shut down Alexa”, and in the log file repeated timed log entries of the form:

 “2020-01-07 21:50:12,185 : [INFO] (Alexa     ) Critical exception n°7: ('str() takes at most 1 argument (2 given)',)”
(screen shots attached)

Can anyone suggest what the problem is and how to fix it?

Note I am close to being a complete python/Raspberry Pi novice, although generally fairly IT competent.

Thanks in advance

[image]
[image]

MichaelB2018, 2020-01-07:
Afraid we have multiple problems here. So first, I need to get to the real error message. To do so, can you edit line 91 of myalexa.py. It currently reads:

`                self.LogInfo("Critical exception nÂ°" + str(error) + ": "+ str(e.args))`

and it should read (note the "[0]"):

`                self.LogInfo("Critical exception nÂ°" + str(error) + ": "+ str(e.args[0]))`

Once changed, can you re-run and post me the real error message. Once we got to the bottom of the problem, I'll check in a code change to fix the above issue too.

nstone33, 2020-01-07:
Thanks for your prompt reply. As I’m a novice on python, could you give me a few pointers on how to edit the file?

MichaelB2018, 2020-01-07:
1.) I assume you were able to find the file: myalexa.py. So go to that directory
2.) type: nano myalexa.py
3.) type: CTRL-_ (so press the Control Key and type "_" (ie underscore). Then type 91
4.) simply move the cursor and edit the line
5.) Type CTRL-X to exit and then type "Y" to save.

MichaelB2018, 2020-01-07:
@rgtrex: Reviewing your issue theoretically (as I only have a Echo Dot 1st gen, not 3rd gen), I could not find anything glaring. I found a few likes on the internet about problems with fauxmo and Echo Dot 3rd gen, but non of them seems to describe a problem that we may face. SO then I reviewed the latest version of FauxMo (https://github.com/n8henrie/fauxmo) and again, I could not find anything glaring. 

So, afraid I will need your help to test this out. The easiest first step is probably for you to install the latest fauxmo version out of the box and see if we can get that to work. You can go to https://github.com/n8henrie/fauxmo and follow those instructions. If that works, we know that any recent change has caused our problem and at least we then can work one by one through it.

https://github.com/n8henrie/fauxmo

nstone33, 2020-01-08:
**MichaelB2018**: I have edited the file  and restarted Pi-Somfy, and attached screenshots of the edited file and log (looks the same??).

Really appreciate your help. As it’s gone midnight where I am, I’ll respond to any further advice from later on in the day,
[image]
[image]

 after work.

MichaelB2018, 2020-01-08:
thanks - but I'm afraid it did not uncover the problem as you probably guessed... Let's do the following. Edit the same file again, but just after line 91, insert the following:

`traceback.print_exc()`

This hopefully should give the full details of the exception.

rgtrex, 2020-01-08:
> I have got blind control to work via manual control on the web interface but been unable to get Alexa to recognise the two blinds. After the first few failures with Alexa I removed one of them from [...]

I guess you installed "SOMFY-PI" project Step-By-Step according to instructions?!
If you did so,... I also guess that you used Python 3. therefore you need to run the following line in cording to start Web interface and Alexa interface:
sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e

The differences between "my" command and the project command is the "3" number after "python"

I really don't know why it's working in that way,

If this works for you and you get a connection to Alexa. please update us, how you did so.

rgtrex, 2020-01-08:
@MichaelB2018 Thanks for your reply
Can you please simplify the following sentence to me:
"[[4. Make a config.json based on config-sample.json]](https://github.com/n8henrie/fauxmo#simple-install-from-pypi)" Step 4 in the procedure.

Do I need to install all?
**"Simple install: From PyPI"
"Simple install of dev branch from GitHub"
"Install for development from GitHub"**

Thanks again.

nstone33, 2020-01-08:
**@rgtrex:** Thanks for your advice to change python to python3. Didn't work the first time (may not have started Pi-Somfy properly as I got no response) but I ran the command again and then Alexa discovered both devices together.  Now working fine. 

**@MichaelB2018:** As you can see from my comment above, I am sorted.  Appreciate your help yesterday.

rgtrex, 2020-01-08:
@nstone33 I'm happy for you.
Are you using Raspberry Pi 4 and Echo Dot 3?

Nickduino, 2020-01-08:
Shit, I should have tilted earlier: of course fauxmo doesn't like Python 2.7

rgtrex, 2020-01-08:
@Nickduino Hi, can you please try to help me with my issue:
copied form my previous comment:
Whenever I ask Alexa to Discover Devices I get the following lines:

**2019-12-31 06:59:33,260 : [DEBUG] (Alexa ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 10.100.102.10:54337\r\nAccept: /\r\nContent-Type: application/json\r\n\r\n'
2019-12-31 06:59:33,260 : [INFO] (Alexa ) Responding to setup.xml for Kitchen Shutter
2019-12-31 06:59:34,294 : [DEBUG] (Alexa ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 10.100.102.10:54338\r\nAccept: /\r\nContent-Type: application/json\r\n\r\n'
2019-12-31 06:59:34,295 : [INFO] (Alexa ) Responding to setup.xml for Side Balcony Shutter**

Alexa seems to be able to contact Raspberry Pi but Raspberry Pi does not "answer" her. And of course Alexa doesn't find any device (shutter). Makes sense?

More info:
Raspberry Pi 4
Echo Dot 3rd

Maybe port issue? or python version issue?
Fauxmo.py- File not suitable for working with Echo Dot 3rd?

What can I do to solve this problem?

MichaelB2018, 2020-01-08:
@rgtrex: you state "Alexa seems to be able to contact Raspberry Pi but Raspberry Pi does not "answer" her. " I suspect that's wrong. Much more likely, the RPi is indeed responding to Alexa, but Alexa does not understand the response. You can see the response in fauxmo.py, line 50-73. And that brings me to the version of Echo Dot you run. I got a 1st gen Echo dot, but you have a 3rd gen one. And I suspect this to be the problem and hence I can't help you to debug. See my response from yesterday above (copied here):

Reviewing your issue theoretically (as I only have a Echo Dot 1st gen, not 3rd gen), I could not find anything glaring. I found a few links on the internet about problems with fauxmo and Echo Dot 3rd gen, but non of them seems to describe a problem that we may face. So then I reviewed the latest version of FauxMo (https://github.com/n8henrie/fauxmo) and again, I could not find anything glaring.

Bottom line, afraid I will need your help to test this out. The easiest first step is probably for you to install the latest fauxmo version out of the box and see if we can get that to work. You can go to https://github.com/n8henrie/fauxmo and follow those instructions. If that works, we know that any recent change has caused our problem and at least we then can work one by one through it.

nstone33, 2020-01-09:
**@rgtrex** : Raspberry Pi 3 and Echo Dot 3rd Gen. I’m interested that you ask about what Alexa devices I have, as I assumed the main interaction between Pi- Somfy and Alexa is with the Alexa system servers, with the home devices just conduits for that communication. I also have a “previous generation” Echo which I assume also works with Pi .Somfy although I realise I haven’t actually tried it yet.

lfliess, 2020-01-09:
works with me echo dot 3.
the only issue I had back then was running with python 2
python 3 worked

MichaelB2018, 2020-01-10:
so we know that Echo Dot 3rd gen works in general. Then let's go back to @rgtrex original suggestion and make sure RPi responds to Alexa. 

@rgtrex, can edit fauxmo.py and un-comment line 246. That should show us the actual message sent back to Alexa.

rgtrex, 2020-01-10:
@MichaelB2018 

I "enabled" the comment at line 246 and I got this:

responsed to setup-->HTTP/1.1 200 OK
CONTENT-LENGTH: 1120
CONTENT-TYPE: text/xml
DATE: Fri, 10 Jan 2020 15:27:44 GMT
LAST-MODIFIED: Sat, 01 Jan 2000 00:01:15 GMT
SERVER: Unspecified, UPnP/1.0, Unspecified
X-User-Agent: redsonic
CONNECTION: close

<?xml version=1.0?>
            <root>
             <device>
                <deviceType>urn:Belkin:device:controllee:1</deviceType>
                <friendlyName>Shutter</friendlyName>
                <manufacturer>Belkin International Inc.</manufacturer>
                <modelName>Socket</modelName>
                <modelNumber>3.1415</modelNumber>
                <modelDescription>Belkin Plugin Socket 1.0</modelDescription>

                <UDN>uuid:Socket-1_0-2ef53687574746</UDN>
                <serialNumber>221517K0101769</serialNumber>
                <binaryState>0</binaryState>
                <serviceList>
                  <service>
                      <serviceType>urn:Belkin:service:basicevent:1</serviceType>
                      <serviceId>urn:Belkin:serviceId:basicevent1</serviceId>
                      <controlURL>/upnp/control/basicevent1</controlURL>
                      <eventSubURL>/upnp/event/basicevent1</eventSubURL>
                      <SCPDURL>/eventservice.xml</SCPDURL>
                  </service>
              </serviceList> 
              </device>
            </root>

In the log file I got:
**2020-01-10 17:27:44,746 : [DEBUG] (Alexa     ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 10.100.102.22:54337\r\nAccept: */*\r\nContent-Type: application/json\r\n\r\n'
2020-01-10 17:27:44,747 : [INFO] (Alexa     ) Responding to setup.xml for Shutter**

MichaelB2018, 2020-01-11:
This does look like a correct response to me, really not sure why your Echo Dot 3rd gen does not seem to accept it. Do you have any thought what may be different in your set up than for @lfliess ? The only other thought I have is to try it with native fauxmo as per previous response. But that's throwing the kitchen sink to be honest.

rgtrex, 2020-01-12:
@MichaelB2018 
I can only guess what can be different:

1. I'm using Raspberry PI 4, maybe @lfliess is using RP Ver. 3B.
2. Maybe I need to ENABLE/DISABLE features in Raspberry PI 4 before or after installation?
3. Maybe my Raspberry PI 4 is "naked" from projects and software installation, therefore I need to install pre-required software/services/updates etc.?
4. Maybe I need to install/enable any feature at my Alexa Developer Account? or something like that?  

I notice that a few of those who have this project and its work for them with ECHO 3 also have ECHO 2 at the same network.

I tried to run this project with "NOOBS version 3.2.1" and with "Raspbian Buster with desktop and recommended software. version September 2019" and I end up with the same results....not working with Alexa.(I got the same logs in both cases)

By the way is there any chance that **this** project can run with MQTT via **SMARTTHINGS hub**?
I tried to find a step by step guide to do this, but I couldn't find one.

MichaelB2018, 2020-01-12:
It does support MQTT, but I'm not  familiar with Smartthings Hub.

As for the problem, I still think  it may be linked to Alexa, not to the RPi. (I have a RPi 3 though). @lfliess: do you have both an Echo Dot 2nd and 3rd gen? If so, that may be the problem. It's at least possible that the Echo Dot 2nd get acts as a hub.

closed 2020-04-03 by Nickduino

Nickduino, 2020-08-24:
Ideas:
- Restart your Echo device
- Use Wireshark or equivalent to listen to traffic. There's a great topic about that on the fauxmo repo

kjellhult, 2020-08-24:
Thanks for replying.  I have rebooted my Echo several times with no change in behavior.

I originally had a smart Kwikset deadbolt lock enabled on the Echo, but once any security equipment has been added to the Echo, one must use the Amazon Alexa app to add any additional items (Alexa says to do that when asked to scan for new devices).  I was originally trying to locate the Pi-Somfy using my Alexa app on my phone but it never found anything.

That said, I *did* delete the lock from the Echo and try to scan from there - again, no change in behavior.  (This is before and after rebooting the Echo Plus second gen).

I will work with wireshark - supposing I will look at traffic between the Pi and my phone while trying to add it?  Or should I scrap the lock again and use the Echo Plus?

Is there any benefit to pairing the Pi over ethernet compared to wifi?  RIght now on wifi but easy to change if necessary.

Again, thank you for your help.  My wife loves controlling our outdoor patio awning from her phone; if she can yell at Alexa and do the same we all look like heroes.

Nickduino, 2020-08-25:
> I will work with wireshark - supposing I will look at traffic between the Pi and my phone while trying to add it? 

I would do that. Your router should broadcast a message from the Echo to every device connected. See if the Pi responds. So, ideally, wireshark will run on your Pi

kjellhult, 2020-08-30:
I'm thinking that perhaps the Alexa service is not starting properly because the interface hangs with the following command:

```
pi@RasPi3BWorkerBee:~ $ sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e
Creating new config file : /home/pi/Pi-Somfy/operateShutters.conf
/home/pi/Pi-Somfy/defaultConfig.conf
 * Serving Flask app "WebServer" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off

        <------- here's where I need to hit control-C because it appears to hang -------->

^CTraceback (most recent call last):
  File "/home/pi/Pi-Somfy/operateShutters.py", line 557, in <module>
    MyShutter = operateShutters(args = args)
  File "/home/pi/Pi-Somfy/operateShutters.py", line 392, in __init__
    self.ProcessCommand(args);
  File "/home/pi/Pi-Somfy/operateShutters.py", line 490, in ProcessCommand
    self.alexa.setDaemon(True)
  File "/usr/lib/python3.7/threading.py", line 1132, in setDaemon
    self.daemon = daemonic
  File "/usr/lib/python3.7/threading.py", line 1125, in daemon
    raise RuntimeError("cannot set daemon status of active thread")
RuntimeError: cannot set daemon status of active thread
pi@RasPi3BWorkerBee:~ $
```

This is happening with a brand-new install.  Literally started from scratch with a newly-flashed SD card and nothing else installed.  Does anyone else have an issue with that first command causing a hang?  If I then start the service and retype the command it behaves normally:

```
pi@RasPi3BWorkerBee:~ $ sudo bash /home/pi/Pi-Somfy/installService.sh 
Created symlink /etc/systemd/system/multi-user.target.wants/shutters.service → /etc/systemd/system/shutters.service.
pi@RasPi3BWorkerBee:~ $ sudo systemctl start shutters.service
pi@RasPi3BWorkerBee:~ $
pi@RasPi3BWorkerBee:/var/log $ sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e
pi@RasPi3BWorkerBee:/var/log $
```

The awning works via the web interface well, but Alexa doesn't detect any new devices when I scan.  Per my post above, I am scanning from my Alexa app on my phone since my Echo already has locks added, and will not scan for new devices if security hardware is connected.  (I deleted my lock from the system and tried from my Echo Plus with no luck - no change in behavior).

Wireshark logs to follow...

kjellhult, 2020-08-30:
Here's the tshark output.  Tshark installed on the RasPi 3B+ (192.168.3.104) --

```
pi@RasPi3BWorkerBee:~ $ sudo tshark -f "host 192.168.3.104" -Y "udp"
Running as user "root" and group "root". This could be dangerous.
Capturing on 'wlan0'
    5 102.252797173 192.168.3.104 → 192.168.3.146 UDP 426 46505 → 32768 Len=384
   11 138.322238168 192.168.3.104 → 192.168.3.170 MDNS 122 Standard query response 0x0365 PTR 104.3.168.192.in-addr.arpa, "QM" question PTR RasPi3BWorkerBee.local
^C2 packets captured
pi@RasPi3BWorkerBee:~ $ 

```

192.168.3.146 is our Sonos system (not sure why they're talking to each other) and 192.168.3.170 is my laptop -- I used Angry IP scanner to find out what x.x.x.146 was in the middle of the process.  No other traffic detected in the 15 minutes or so.  And during that time I ran the "discover new devices" routine in the Alexa app from both my Android phone and my iPad.

Nothing...

Here's the very long unredacted shutters log:

```
pi@RasPi3BWorkerBee:~ $ cat /var/log/operateShutters.log
2020-08-30 21:32:15,529 : [INFO] (Thread-72 ) processing Command "deleteShutter" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('id', '0x279622')])])
2020-08-30 21:32:15,530 : [INFO] (Thread-73 ) processing Command "getConfig" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([])])
2020-08-30 21:32:15,533 : [DEBUG] (Thread-72 ) delete shutter: 0x279622
2020-08-30 21:32:15,534 : [DEBUG] (Thread-73 ) getConfig called, sending: {"Latitude": 34.05265942137599, "Longitude": -117.90527343750001, "Schedule": {}, "ShutterDurations": {"0x279621": 20, "0x279622": "15"}, "Shutters": {"0x279621": "awning", "0x279622": "test"}}
2020-08-30 21:32:15,541 : [DEBUG] (Thread-72 ) CONFIG FILE WRITE ->> mySectionStart = 88, mySectionEnd = 90, myLine = 90
2020-08-30 21:32:17,087 : [DEBUG] (Thread-74 ) http://192.168.3.104/cmd/getConfig ( GET ): ((), {'command': 'getConfig'}) | {}
2020-08-30 21:32:17,093 : [INFO] (Thread-74 ) processing Command "getConfig" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([])])
2020-08-30 21:32:17,109 : [DEBUG] (Thread-74 ) getConfig called, sending: {"Latitude": 34.05265942137599, "Longitude": -117.90527343750001, "Schedule": {}, "ShutterDurations": {"0x279621": 20}, "Shutters": {"0x279621": "awning"}}
2020-08-30 21:33:02,772 : [DEBUG] (Thread-81 ) http://192.168.3.104/cmd/up ( POST ): ((), {'command': 'up'}) | {}
2020-08-30 21:33:02,775 : [INFO] (Thread-81 ) processing Command "up" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279621')])])
2020-08-30 21:33:02,775 : [DEBUG] (Thread-81 ) rise shutter "0x279621"
2020-08-30 21:33:02,776 : [INFO] (Thread-81 ) [awning] Going up
2020-08-30 21:33:02,777 : [DEBUG] (Thread-81 ) sendCommand: Waiting for Lock
2020-08-30 21:33:02,777 : [DEBUG] (Thread-81 ) sendCommand: Lock aquired
2020-08-30 21:33:02,780 : [DEBUG] (Thread-81 ) CONFIG FILE WRITE ->> mySectionStart = 97, mySectionEnd = 99, myLine = 98
2020-08-30 21:33:02,793 : [INFO] (Thread-81 ) Remote  :      0x279621 (awning)
2020-08-30 21:33:02,793 : [INFO] (Thread-81 ) Button  :      0x02
2020-08-30 21:33:02,794 : [INFO] (Thread-81 ) Rolling code : 15
2020-08-30 21:33:02,795 : [INFO] (Thread-81 ) 
2020-08-30 21:33:02,795 : [INFO] (Thread-81 ) Frame  :    0xA7 0x20 0x00 0x0F 0x27 0x96 0x21 
2020-08-30 21:33:02,796 : [INFO] (Thread-81 ) With cks  : 0xA7 0x29 0x00 0x0F 0x27 0x96 0x21 
2020-08-30 21:33:02,797 : [INFO] (Thread-81 ) Obfuscated :0xA7 0x8E 0x8E 0x81 0xA6 0x30 0x11 
2020-08-30 21:33:03,413 : [DEBUG] (Thread-81 ) sendCommand: Lock released
2020-08-30 21:33:03,415 : [DEBUG] (Thread-83 ) [awning] Waiting for operation to complete for 20.0 seconds
2020-08-30 21:33:07,535 : [DEBUG] (Thread-84 ) http://192.168.3.104/cmd/stop ( POST ): ((), {'command': 'stop'}) | {}
2020-08-30 21:33:07,538 : [INFO] (Thread-84 ) processing Command "stop" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279621')])])
2020-08-30 21:33:07,539 : [DEBUG] (Thread-84 ) stop shutter "0x279621"
2020-08-30 21:33:07,540 : [INFO] (Thread-84 ) [awning] Stopping
2020-08-30 21:33:07,541 : [DEBUG] (Thread-84 ) sendCommand: Waiting for Lock
2020-08-30 21:33:07,541 : [DEBUG] (Thread-84 ) sendCommand: Lock aquired
2020-08-30 21:33:07,544 : [DEBUG] (Thread-84 ) CONFIG FILE WRITE ->> mySectionStart = 97, mySectionEnd = 99, myLine = 98
2020-08-30 21:33:07,556 : [INFO] (Thread-84 ) Remote  :      0x279621 (awning)
2020-08-30 21:33:07,557 : [INFO] (Thread-84 ) Button  :      0x01
2020-08-30 21:33:07,558 : [INFO] (Thread-84 ) Rolling code : 16
2020-08-30 21:33:07,558 : [INFO] (Thread-84 ) 
2020-08-30 21:33:07,559 : [INFO] (Thread-84 ) Frame  :    0xA7 0x10 0x00 0x10 0x27 0x96 0x21 
2020-08-30 21:33:07,560 : [INFO] (Thread-84 ) With cks  : 0xA7 0x14 0x00 0x10 0x27 0x96 0x21 
2020-08-30 21:33:07,560 : [INFO] (Thread-84 ) Obfuscated :0xA7 0xB3 0xB3 0xA3 0x84 0x12 0x33 
2020-08-30 21:33:08,177 : [DEBUG] (Thread-84 ) sendCommand: Lock released
2020-08-30 21:33:08,178 : [DEBUG] (Thread-84 ) [0x279621] Previous position: 0
2020-08-30 21:33:08,188 : [DEBUG] (Thread-84 ) [0x279621] Seconds since last command: 5
2020-08-30 21:33:08,189 : [DEBUG] (Thread-84 ) [0x279621] Duration percentage: 25, State position: 0
2020-08-30 21:33:13,982 : [DEBUG] (Thread-85 ) http://192.168.3.104/
[... cut ...]
2020-08-30 22:06:51,378 : [DEBUG] (MainThread) Loading Schedule from Config File
2020-08-30 22:06:51,385 : [INFO] (Scheduler ) Today is 2020/08/30, a Sun, Sunrise is at 14:23:59.167158 and Sunset is at 03:20:47.313462
2020-08-30 22:06:51,385 : [INFO] (Alexa     ) Entering fauxmo polling loop
2020-08-30 22:06:51,386 : [INFO] (MQTT      ) Entering MQTT polling loop
2020-08-30 22:06:51,386 : [DEBUG] (Scheduler ) {}
2020-08-30 22:06:51,407 : [INFO] (MainThread) Starting WebServer on Port 80
pi@RasPi3BWorkerBee:~ $ 
```
Any ideas on a next step?  Thank you very much.

cm-mojo, 2020-11-05:
anyone get any further with this? i'm having the same problem.