Nickduino/Pi-Somfy #49: How to test if the emitter works?

MatthiasGrandl, 2020-01-09:
I setup the software properly on my PI. Can't get my Awning (Markiluxx using Somfy) to pair though, even after putting it in programming mode. I have no clue whether or not the emitter works in the first place. Would appreciate any pointers.

ttimpe, 2020-04-02:
Kind of the same situation here, I soldered the right crystal onto a FS 1000A however I am unable to get any response from my shutters.

Nickduino, 2020-04-03:
I'd say either use the algorithm (or my Arduino version) to create a Somfy decoder or check your wiring 10 times

Ezezi3l, 2020-04-16:
it is very easy to test the emitter with a SDR dongle and SDR Sharp software.

MichaelB2018, 2020-04-18:
could you post some detailed instructions on how to test an emitter, many people (like me) may find this useful.

Ezezi3l, 2020-04-20:
- you need a simple SDR usb receiver like this one: https://www.adafruit.com/product/1497
- install and setup SDRSharp from: https://airspy.com/download/
- follow instruction to install the correct usb driver
In SDRSharp listen to the frequency 443.42Mhz.
Send an instruction to one shutter and you should see a short communication on 443.42Mhz
[image]

StarScream159, 2020-05-29:
This would work for testing right? https://www.amazon.ca/digital-Stick-RTL-SDR-Receiver-RTL2832U/dp/B00CM3LNMM/

Nickduino, 2020-06-25:
@Ezezi3l Can you also read the data sent by the remote (in binary or hex) or is it just a spectrum analyser?

rbswift, 2020-06-25:
This might be a good place to start... https://pushstack.wordpress.com/2014/05/17/somfy-rtl-sdr/

Nickduino, 2020-06-30:
Good! That's the guy who originally reverse engineered the protocol

kjellhult, 2020-08-30:
In terms of the hardware -- when you replace the crystal, keep in mind that one of the three pins is grounded to the metal case and the others have *very small* insulators around their pads.  Make sure you clean up any excess solder and consider a small air gap between the emitter and the board to make sure the other two pins don't short to ground.  I was having intermittent issues until I re-soldered the oscillator with a tiny gap between the board and the case.

StarScream159, 2020-09-22:
@kjellhult can you post a really close up picture of how you soldered the oscillator?

kjellhult, 2020-09-24:
[image]
I've attached a close-up photo of the oscillator kind of floating off the surface of the board to prevent any bleed-through solder from the pins other than the ground pin from shorting the metal case to another pin.  With any luck this helps - it certainly was an issue for me with my crappy soldering skills.

closed 2021-10-10 by Nickduino