Nickduino/Pi-Somfy #50: Running Pi-Somfy from another Python script linked to heating control

nstone33, 2020-01-18:
With help from others on this project I have my blinds controlled by Pi-Somfy and Alexa.  I would now like to link this to information from my central heating program (tado) so that when everyone is out of the house the blinds will close (perhaps also subject to other conditions, but that's for another time).  I have been able to access data from tado and download and save it in a text file on my Raspberry Pi, and create a Python script that will identify if the string that indicates whether the occupants are in  HOME or AWAY). I now want to use this IF statement to trigger Pi-Somfy to close my two blind, but this is where I am struggling, mainly because I am a virtual Python beginner.

The script I have below (HOME would be replaced with AWAY in the final version) checks if the word HOME is in the text file and then if so prints the text and confirms "True":

#!/usr/bin/python3
w = ("HOME") # For Python 3: use input() instead
with open('/home/pi/libtado/tadostate.txt') as f:
    found = False
    for line in f:
        if w in line: # Key line: check if `w` is in the line.
            print(line)
            found = True
            print  (found)
    if not found:
        print('The word cannot be found!')

I have used various web resources to help me figure out the syntax I need to run the Pi-Somfy instruction to close a blind (replacing the three lines currently within the IF statement), but all attempts have not worked.  I'm think that in part this is due to incorrect file structure and/or paths (relevant part of directory tree below)

  ├── myscheduler.pyc
│   ├── mywebserver.py
│   ├── mywebserver.pyc
│   ├── operateShutters.conf
│   ├── operateShutters.py
│   ├── __pycache__
│   │   ├── fauxmo.cpython-37.pyc
│   │   ├── myalexa.cpython-37.pyc
│   │   ├── myconfig.cpython-37.pyc
│   │   ├── mylog.cpython-37.pyc
│   │   ├── mymqtt.cpython-37.pyc
│   │   ├── myscheduler.cpython-37.pyc
│   │   └── mywebserver.cpython-37.pyc
│   └── README.md
├── Public
├── tado
├── tadoquery
│   ├── __init__.py
│   └── tadoquery4.py
├── tadoquery2.py
├── tadoquery3.py
├── tadoquery.py
├── Templates
└── Videos

Currently I have the above script (tadoquery4.py) in a home/pi/tadoquery directory together with a blanks __init__.py file and have used the line 'from .import operateShutters'  really just to see if would recognise the operateShutters.py, but this returns 'cannot import name 'operateShutters' from '__main__' (/home/pi/tadoquery/tadoquery4.py)'. I have previously had the tadoquery4.py file in the home/pi directory  (ie the same directory as operateShutters.py) but that also did not work.  Of course, import may be competely the wrong method for what I am trying to do!

So I am unclear how to run the operateShutters.py script from within my tadoquery4.py script and how to specify the parameters (blind name and open or close).

I would be very grateful for some help on this.

MichaelB2018, 2020-01-18:
May I suggest you use Home Assistant (https://www.home-assistant.io/) It supports both tado and Pi-Somfy and allows for such integrations in a much easier way. Alternatively you could write a small Python program that sends some basic messages to Pi-Somfy over MQTT whenever the state of the blinds need to be changes. U use mosquitto for MQTT (https://mosquitto.org/). you can then send simple messages such as 

```
somfy/0x2670xx/level/cmd 0
somfy/0x2670xx/level/cmd STOP
somfy/0x2670xx/level/cmd 100
```

nstone33, 2020-01-19:
Thanks Michael.  I had a look at Home Assistant but I couldn't see a
reference to Pi-Somfy as supported.  I also looked up Mosquitto and
installed it. Could you give me some pointers on how I could set up Home
Assistant and Pi-Somfy to integrate with tado? Or mosquitto?  Which would
be easier for a real novice like me?

Very many thanks
Neil

MichaelB2018, 2020-01-19:
1.) Start operateShutters.py with the -m option. should look somethign like this:

`operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -e -m`

2.) assuming you set up mosquitto with a username and password, you can use the following command to listen to the message bus and see what's going on:

`mosquitto_sub -h 192.168.x.x -p 1883 -u [username]-P [password] -t '#' -v`

3.) now to post a message to MQTT, use

`mosquitto_pub -h 192.168.x.x -p 1883 -u [username]-P [password] -t 'somfy/0x2670xx/level/cmd' -m '0'`
`mosquitto_pub -h 192.168.x.x -p 1883 -u [username]-P [password] -t 'somfy/0x2670xx/level/cmd' -m '100'`

So that should operate the blind. 

4.) Now in write a small python program that posts a message based on TADO's output: 
you can use the user guide from here:

http://www.steves-internet-guide.com/publishing-messages-mqtt-client/

or you just write a shell script that calls the above 2 commands

lfliess, 2020-01-19:
I would still go with home assistant. it would allow you to add more rules and more integrations and have everything work together.

Pi-Somfy is not directly supported but since Home Assistant has MQTT integration (it can run Mosquito), you will just see your Somfy devices on the dashboard.
after that I would add the Node-RED add on to HA, it allows you to make these rules and integrations without writing a single line of code, just graphical flowcharts.

I installed HA on the same Rpi that Pi-somfy runs on. I can't find the guide now but you're looking for installing hass.IO on a docker on your RPi OS

closed 2020-01-26 by MichaelB2018