Nickduino/Pi-Somfy #51: Using MQTT to control

mgf909, 2020-01-21:
Hi guys - this is awesome Great work. Im able to control, via the WebUI my Somfy shutter without changing the crystal on the 433 transmitter..although i have ordered some!

Anyway , my issue is that i cannot control my blind via MQTT.
As mentioned i can control it fine using the WebUI,  - UP/DOWN/STOP all work very responsively..now i want to integrate it into OpenHAB using MQTT.

Pi-Somfy is communicating with my Mosquito. Its published:
`somfy/0x279621/level/set_state`
with values of 0 and 100

I couldnt see in the README any doco on how to use MQTT, but did find the issue comment here:
https://github.com/Nickduino/Pi-Somfy/issues/50#issuecomment-575934737
showing how to use it...

So i tried publishing :
somfy/0x279621/level/cmd
setting values 0 and 100
i also tried STOP, UP,up,DOWN,down

There is no action from my blind and nothing logged.

What am i doing wrong?

MichaelB2018, 2020-01-21:
Have you started operateShutters.py with the "-m" option?

If so, can you check if you see any issues in the log files:

`grep -i mqtt /var/log/operateShutters.lo*`

Your description is correct and should be workable

mgf909, 2020-01-21:
Hi Michael,
Thanks for your reply.
Yes -m option is used from crontab

`@reboot sleep 60;python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -m
0 * * * * python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a -m

`
There is some activity of mqtt in the log - but i belive this is just Pi-Somfy publishing updates triggered from using the WebUI.
`2020-01-21 02:01:34,501 : [INFO] (Thread-169) 
2020-01-21 02:01:34,504 : [INFO] (Thread-169) Frame  :    0xA7 0x40 0x00 0x17 0x27 0x96 0x21 
2020-01-21 02:01:34,507 : [INFO] (Thread-169) With cks  : 0xA7 0x46 0x00 0x17 0x27 0x96 0x21 
2020-01-21 02:01:34,509 : [INFO] (Thread-169) Obfuscated :0xA7 0xE1 0xE1 0xF6 0xD1 0x47 0x66 
2020-01-21 02:01:34,893 : [DEBUG] (Thread-169) sendCommand: Lock released
2020-01-21 02:01:34,896 : [INFO] (Thread-169) Received request to set Shutter 0x279621 to 0
2020-01-21 02:01:34,898 : [INFO] (Thread-169) sending message to MQTT: somfy/0x279621/level/set_state = 0
2020-01-21 02:01:35,806 : [DEBUG] (Thread-171) http://192.168.2.134/cmd/stop ( POST ): ((), {'command': 'stop'}) | {}
2020-01-21 02:01:35,813 : [INFO] (Thread-171) processing Command "stop" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279621')])])
2020-01-21 02:01:35,816 : [DEBUG] (Thread-171) stop shutter "0x279621"
2020-01-21 02:01:35,818 : [DEBUG] (Thread-171) sendCommand: Waiting for Lock
2020-01-21 02:01:35,821 : [DEBUG] (Thread-171) sendCommand: Lock aquired
2020-01-21 02:01:35,830 : [DEBUG] (Thread-171) CONFIG FILE WRITE ->> mySectionStart = 97, mySectionEnd = 98, myLine = 98
2020-01-21 02:01:35,862 : [INFO] (Thread-171) Remote  :      0x279621 (PoolSide)
2020-01-21 02:01:35,864 : [INFO] (Thread-171) Button  :      0x01
2020-01-21 02:01:35,867 : [INFO] (Thread-171) Rolling code : 24
2020-01-21 02:01:35,869 : [INFO] (Thread-171) 
2020-01-21 02:01:35,872 : [INFO] (Thread-171) F`

mgf909, 2020-01-21:
Fixed - The time/date on my Rpi was incorrect as id not set the correct TZ/Locale for my Pi.
After doing this and rebooting it began working!.

Thanks again Michael and everyone involved with developing this! Excellent!

closed 2020-01-21 by mgf909