Nickduino/Pi-Somfy #54: MQTT /Home Assistant integration

nstone33, 2020-01-28:
I have set up Home Assistant with hass,io and added in MQTT.  I have gone through all the instructions in Pi-Somfy to set up integration with MQTT.  I'm at the stage where I can send a an mosquitto command to open a blind, that shows up if I monitor the messages.  In Home Assistant the MQTT log shows a connection and then a disconnection:

1580244209: New connection from 172.30.32.1 on port 1883.
1580244209: New client connected from 172.30.32.1 as mosqpub|3834-raspberryp (p2, c1, k60, u'neil').
1580244209: Client mosqpub|3834-raspberryp disconnected.

However, the blinds don't respond, and there are no Somfy devices showing in Home Assistant. What should I expect to see in HA, and why no response from the blinds.  Is it normal for the client to connect to MQTT and then disconnect immediately (ie just passing a message and disconnecting) or should the connection to permanent?

Thanks

MichaelB2018, 2020-01-28:
The disconnect is definitely the issue. Could be that he password in operateShutters.conf is wrong or something similar. Suggest you check the log file of the Pi-Somfy:

/var/log/operateShutters.log

Can you see anything in there that indicates why it disconnected again?

nstone33, 2020-01-29:
....I have limited expertise on this, but I can't see anything relating to the disconnection or referring to an incorrect password or similar.

MichaelB2018, 2020-01-29:
Will need to look into this a bit more to understand what might be going on. 

But one more question upfront: If you use the Web GUI to operate the blind (lower it or lift  it), Do you see a message posted on MQTT when it finished? Also do you see a message posted on MQTT when you start operateShutters.py?

nstone33, 2020-01-29:
When I use the web interface I do not see any message on the MQTT log on
hassio when the blind movement finishes.

When starting operateShutters I see this on the MQTT log in hassio:
1580341650: New connection from 172.30.32.1 on port 1883.
[INFO] found neil on Home Assistant
1580341652: New client connected from 172.30.32.1 as
mosqsub|18749-raspberry (p2, c1, k60, u'neil').

Thanks

MichaelB2018, 2020-01-31:
Can you try the following manual command to understand a bit better what goes on the MQTT bus:

1.) assuming you set up mosquitto with a username and password, you can use the following command to listen to the message bus and see what's going on:

mosquitto_sub -h 192.168.x.x -p 1883 -u [username]-P [password] -t '#' -v

2.) now to post a message to MQTT, use (and put in the actual address for your blind, so instead of "0x2670xx", maybe "0x267000"

mosquitto_pub -h 192.168.x.x -p 1883 -u [username]-P [password] -t 'somfy/0x2670xx/level/cmd' -m '0'
mosquitto_pub -h 192.168.x.x -p 1883 -u [username]-P [password] -t 'somfy/0x2670xx/level/cmd' -m '100'

So that should operate the blind. Let me know what comes out.

nstone33, 2020-01-31:
I have a problem at the moment as neither blind is working.  My left blind
stopped being recognised by Alexa and the right one was still ok.  Now
neither work on Alexa or the web interface, but do on the physical remote.
Going to have to reboot the pi and investigate further but no time to do
that now.  (a quick query...don't seems to be able to reuse the same name
for a blind after deleting it from the web interface, or even from the
.conf file. What's the best way to do this?)

Before they stopped working I tried the mqtt commands you suggested on the
right blind, monitoring them using -v.  The commands appeared below the
monitoring request, but the blind did not respond.

Thanks for your help

Dellonb, 2020-12-14:
Seems I have hit the same problem…
I am successfully running Pi-Somfy on separate Raspberry Pi 4 Model B Rev 1.4 2GB for a few weeks now, operating my 4 shutters from Web interface. All works well shutters operation, programming and Scheduling.

Next step is dive HOA.

Recently I have installed Home Assistant 4.16 (core version 0.118.5) on separate Raspberry Pi 4 Model B 4GB and trying to get Pi-Somfy with HOA.

I am stack on Pi-Somfy <> MQTT integration (HOA). All was set up as per given instruction for MQTT but base on troubleshooting Pi-Somfy is not responding to MQTT messages. 
When HOA startas, Mosquitto broker see conection from Pi-Somfy:

1607946106: New connection from 172.30.32.1 on port 1883.
[INFO] found mqtt on Home Assistant
1607946108: New client connected from 172.30.32.1 as 2ss5bhp5FpJTuo6bpIRpZn (p2, c1, k60, u'mqtt').
1607947191: New connection from 172.30.32.1 on port 1883.

First HOA discovery dos not work – I cannot discover my 4 shutters, operateShutters.conf and configuration.yaml configured as per Manual.
There are no Somfy devices showing in Home Assistant.

Secend - I have tested my HOA MQTT broker and all looks OK working with following commands

testing:
mosquitto_sub -d -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t TestTopic -v
mosquitto_pub -d -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t TestTopic -m "OK Works"

When I run below command and operate shutters from Web interface there is nothing showing up on MQTT Broker listen.
mosquitto_sub -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t '#' -v

When I try to run any off following 

Kitchen:
mosquitto_pub -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t 'somfy/0x279621/level/cmd' -m '0'
mosquitto_pub -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t 'somfy/0x279621/level/cmd' -m '100'

Room:
mosquitto_pub -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t 'somfy/0x279622/level/cmd' -m '0'
mosquitto_pub -h 192.168.3.210 -p 1883 -u mqtt -P mqtt2020 -t 'somfy/0x279622/level/cmd' -m '100'

notiong happens to the shutters – I belive MQTT Boker not talking to Pi-Somfy.

I checked log in /var/log/operateShutters.log but there is nothing related to MQTT messages, just regular loging of Web manual action or scheduler.

Any advice?

[image]

rbswift, 2020-12-14:
Post your operateshutters.log file.
Any errors when you ran: sudo pip3 install ephem configparser Flask paho-mqtt ??
Are you using the -mqtt, -m argument?
Post ps -ef

Dellonb, 2020-12-14:
operateshutters.log

2020-12-14 20:01:58,978 : [INFO] (Thread-171) Rolling code : 130
2020-12-14 20:01:58,980 : [INFO] (Thread-171) 
2020-12-14 20:01:58,981 : [INFO] (Thread-171) Frame  :    0xA7 0x10 0x00 0x82 0x27 0x96 0x22 
2020-12-14 20:01:58,981 : [INFO] (Thread-171) With cks  : 0xA7 0x1C 0x00 0x82 0x27 0x96 0x22 
2020-12-14 20:01:58,982 : [INFO] (Thread-171) Obfuscated :0xA7 0xBB 0xBB 0x39 0x1E 0x88 0xAA 
2020-12-14 20:01:59,595 : [DEBUG] (Thread-171) sendCommand: Lock released
2020-12-14 20:01:59,595 : [DEBUG] (Thread-171) [0x279622] Previous position: 0
2020-12-14 20:01:59,596 : [DEBUG] (Thread-171) [0x279622] Seconds since last command: 51
2020-12-14 20:01:59,596 : [WARNING] (Thread-171) [0x279622] Too much time since last command.
2020-12-14 20:01:59,596 : [INFO] (Thread-171) [0x279622] Stay stationary.
2020-12-14 20:02:23,577 : [DEBUG] (Thread-173) http://192.168.3.200/cmd/down ( POST ): ((), {'command': 'down'}) | {}
2020-12-14 20:02:23,579 : [INFO] (Thread-173) processing Command "down" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279622')])])
2020-12-14 20:02:23,580 : [DEBUG] (Thread-173) lower shutter "0x279622"
2020-12-14 20:02:23,580 : [INFO] (Thread-173) [Dining] Going down
2020-12-14 20:02:23,581 : [DEBUG] (Thread-173) sendCommand: Waiting for Lock
2020-12-14 20:02:23,581 : [DEBUG] (Thread-173) sendCommand: Lock aquired
2020-12-14 20:02:23,583 : [DEBUG] (Thread-173) CONFIG FILE WRITE ->> mySectionStart = 99, mySectionEnd = 103, myLine = 101
2020-12-14 20:02:23,593 : [INFO] (Thread-173) Remote  :      0x279622 (Dining)
2020-12-14 20:02:23,593 : [INFO] (Thread-173) Button  :      0x04
2020-12-14 20:02:23,594 : [INFO] (Thread-173) Rolling code : 131
2020-12-14 20:02:23,594 : [INFO] (Thread-173) 
2020-12-14 20:02:23,595 : [INFO] (Thread-173) Frame  :    0xA7 0x40 0x00 0x83 0x27 0x96 0x22 
2020-12-14 20:02:23,595 : [INFO] (Thread-173) With cks  : 0xA7 0x48 0x00 0x83 0x27 0x96 0x22 
2020-12-14 20:02:23,596 : [INFO] (Thread-173) Obfuscated :0xA7 0xEF 0xEF 0x6C 0x4B 0xDD 0xFF 
2020-12-14 20:02:24,209 : [DEBUG] (Thread-173) sendCommand: Lock released
2020-12-14 20:02:24,210 : [DEBUG] (Thread-175) [Dining] Waiting for operation to complete for 0.0 seconds
2020-12-14 20:02:24,211 : [DEBUG] (Thread-175) [Dining] Set new final position: 0
2020-12-14 20:50:07,972 : [INFO] (MainThread) pigpiod was not running
2020-12-14 20:50:08,812 : [INFO] (MainThread) pigpiod is running, process ID is 598 
2020-12-14 20:50:08,816 : [INFO] (MainThread) pigpio's pi instantiated
2020-12-14 20:50:08,817 : [WARNING] (MainThread) WARNING: Failed to join multicast group:
2020-12-14 20:50:08,818 : [INFO] (MainThread) Remote address in dec: 2594337, WeMo port will be n°54337
2020-12-14 20:50:08,819 : [INFO] (MainThread) UPnP broadcast listener: new device registered
2020-12-14 20:50:08,819 : [INFO] (MainThread) FauxMo device 'Kitchen' ready on 127.0.0.1:54337
2020-12-14 20:50:08,819 : [INFO] (MainThread) Remote address in dec: 2594338, WeMo port will be n°54338
2020-12-14 20:50:08,820 : [INFO] (MainThread) UPnP broadcast listener: new device registered
2020-12-14 20:50:08,820 : [INFO] (MainThread) FauxMo device 'Dining' ready on 127.0.0.1:54338
2020-12-14 20:50:08,820 : [INFO] (MainThread) Remote address in dec: 2594339, WeMo port will be n°54339
2020-12-14 20:50:08,821 : [INFO] (MainThread) UPnP broadcast listener: new device registered
2020-12-14 20:50:08,821 : [INFO] (MainThread) FauxMo device 'Terrace' ready on 127.0.0.1:54339
2020-12-14 20:50:08,821 : [INFO] (MainThread) Remote address in dec: 2594340, WeMo port will be n°54340
2020-12-14 20:50:08,821 : [INFO] (MainThread) UPnP broadcast listener: new device registered
2020-12-14 20:50:08,822 : [INFO] (MainThread) FauxMo device 'Salon' ready on 127.0.0.1:54340
2020-12-14 20:50:08,822 : [DEBUG] (MainThread) Loading Schedule from Config File
2020-12-14 20:50:08,822 : [DEBUG] (MainThread) Loading Scheudle 5
2020-12-14 20:50:08,823 : [DEBUG] (MainThread) addEvent: Waiting for Lock
2020-12-14 20:50:08,823 : [DEBUG] (MainThread) addEvent: Lock aquired
2020-12-14 20:50:08,823 : [DEBUG] (MainThread) addEvent: Lock released
2020-12-14 20:50:08,824 : [DEBUG] (MainThread) Loading Scheudle 6
2020-12-14 20:50:08,824 : [DEBUG] (MainThread) addEvent: Waiting for Lock
2020-12-14 20:50:08,824 : [DEBUG] (MainThread) addEvent: Lock aquired
2020-12-14 20:50:08,824 : [DEBUG] (MainThread) addEvent: Lock released
2020-12-14 20:50:08,824 : [DEBUG] (MainThread) Loading Scheudle 7
2020-12-14 20:50:08,825 : [DEBUG] (MainThread) addEvent: Waiting for Lock
2020-12-14 20:50:08,825 : [DEBUG] (MainThread) addEvent: Lock aquired
2020-12-14 20:50:08,825 : [DEBUG] (MainThread) addEvent: Lock released
2020-12-14 20:50:08,825 : [DEBUG] (MainThread) Loading Scheudle 8
2020-12-14 20:50:08,825 : [DEBUG] (MainThread) addEvent: Waiting for Lock
2020-12-14 20:50:08,826 : [DEBUG] (MainThread) addEvent: Lock aquired
2020-12-14 20:50:08,826 : [DEBUG] (MainThread) addEvent: Lock released
2020-12-14 20:50:08,826 : [DEBUG] (MainThread) Loading Scheudle 9
2020-12-14 20:50:08,835 : [DEBUG] (MainThread) addEvent: Waiting for Lock
2020-12-14 20:50:08,835 : [DEBUG] (MainThread) addEvent: Lock aquired
2020-12-14 20:50:08,835 : [DEBUG] (MainThread) addEvent: Lock released
2020-12-14 20:50:08,835 : [DEBUG] (MainThread) Loading Scheudle 10
2020-12-14 20:50:08,836 : [DEBUG] (MainThread) addEvent: Waiting for Lock
[... cut ...]
root      1003     2  0 21:35 ?        00:00:00 [kworker/u8:0-events_unbound]
root      1014     2  0 21:35 ?        00:00:00 [kworker/1:0-mm_percpu_wq]
root      1015     2  0 21:35 ?        00:00:00 [kworker/3:0-events]
root      1023     2  0 21:38 ?        00:00:00 [kworker/0:2]
root      1055     2  0 21:39 ?        00:00:00 [kworker/u8:3]
root      1056     2  0 21:40 ?        00:00:00 [kworker/1:2-mm_percpu_wq]
root      1069     2  0 21:41 ?        00:00:00 [kworker/3:2-mm_percpu_wq]
pi        1070  1002  0 21:42 pts/0    00:00:00 ps -ef
pi@raspberrypi4-2G:~ $

rbswift, 2020-12-14:
`Connected to MQTT with result code 5`
Error 5 is Connection refused – not authorised
Maybe you have a mismatched username and password between your mqtt server and your Pi-Somfy config. See section 7 of the documentation for how to specify `MQTT_User` and `MQTT_Password`

rbswift, 2020-12-14:
Also your `ps -ef `didn't copy/paste properly so the end of the relevant lines is missing but you might want to check why `operateshutters.py` is running twice (PIDs 338 and 426) ... mqtt authentication is likely the main problem here still.

Dellonb, 2020-12-15:
"Error 5 is Connection refused – not authorised" - exactly this was the problem. 

My “operateShutters.conf” file  had wrong User=mq**q**t where created User=mq**t**t  - just stupid typo qq/tt
I can operate shutters now with mqtt commands
Naturally when mqtt works my 4 Shutters were automatic Discovered.

All work now
Thanks for great support

Nickduino, 2021-10-10:
> All work now

Great, thanks @rbswift

closed 2021-10-10 by Nickduino