Nickduino/Pi-Somfy #55: How to reverse remote buttons

geokscott, 2020-03-14:
First of all, thank you for sharing your awesome work on this project!!!

This is exactly what I needed and saved me the pain of trying to figure it out.  I have a Sunsetter Awning that has a Somfy remote control built in.  I also run Home Assistant on a Pi (Hassbian) and got this up and running in no time at all.  Great Read Me.  

I have one issue with my Awning.  I reverses the remote button function from "normal".  Even though it's possible to reverse remote button operation programmatically between the remote and receiver, the Sunsetter Awning will not function properly with them reversed, it's strictly prohibited for some reason.  So my Home Assistant UI shows the Awning OPEN all the time unless it's in the full open position, then it registers as CLOSED.  

I would like to reverse this behavior in this code.  I think I might be able to configure some Cover Template in Home Assistant to revers it, but I'm not positive.  I would rather mod this code to reverse the behavior at the source, rather than programming around the issue in Home Assistant.  Maybe this could be a new option in the conf file?  Could you point me to all the places I should touch to mod the code?

Thanks!

geokscott, 2020-03-15:
Well after OVER thinking the mqtt code for several hours, it turnes out to be much simpler than I originally thought.  All I had to do was reverse the buttons in operateShutters.py
    buttonUp = 0x4
    buttonDown = 0x2

Nickduino, 2020-03-15:
Glad you found your workaround!

closed 2020-03-15 by Nickduino