Nickduino/Pi-Somfy #56: additional antenna

flayfl, 2020-03-18:
Hi at no. 5 
Solder an antenna to the ANT pad :
the link points to a 1 wire antenna ... do i need to bend it and solder both ends as in the image  or it's ok one end only?
a bit confused  .. 
by the way thanks for the repo & instructions.
Flavius

Nickduino, 2020-03-18:
What you see on the image isn't the antenna but a coil (that was on the emitter out of the box)

The antenna : solder only one end of your 17cm wire. It can be straight or helical, it shouldn't matter too much.

flayfl, 2020-03-18:
k the one I've got didn't had that coil .. so bought a helical ... I'll give it a try thanks

Nickduino, 2020-03-18:
You've replaced your oscillator, though, right?

flayfl, 2020-03-20:
yes replaced but seems that it doesn't communicate so far ... i need to find a way to debug the build without an oscilator, i'll dig in over the weekend.

MichaelB2018, 2020-04-18:
see discussion on issue #49

closed 2020-04-18 by MichaelB2018