Nickduino/Pi-Somfy #60: Stop button press sends invalid level to MQTT

rbswift, 2020-04-19:
If I press send a Stop button command to an already stopped blind I get what looks like an invalid level state message via MQTT. For example:

# mosquitto_sub -h localhost -t 'somfy/#' -v
somfy/0x279623/level/set_state 1204
somfy/0x279623/level/set_state 1041

I believe this should be 0-100

MichaelB2018, 2020-04-25:
I think I found the issue and fixed it , but was not able to test it conclusively. Can you test too? Is it working now?

rbswift, 2020-04-25:
I updated and made it fail again once...
`somfy/0x279623/level/set_state 159`
I will try to find some steps to reproduce. It seemed to happen when pressing stop via the web interface whilst the motor is already stopped. My understanding is that this takes the motor to the "my" position which is arbitrarily programmable on each motor and can't possibly be known by the code so it sensibly just guesses to 50%

rbswift, 2020-04-25:
Haven't managed to reproduce it yet.
I do seem to get 4% each time the service is restarted:

`root@pi4:/home/pi/Pi-Somfy# systemctl restart pisomfy.service 
root@pi4:/home/pi/Pi-Somfy# mosquitto_sub -h localhost -t 'somfy/#' v
somfy/0x279623/level/set_state 100
## wait a while then press Stop button on web interface
somfy/0x279623/level/set_state 4
`

Here's the log file for 159% earlier if it's helpful,,,
`2020-04-25 13:20:23,586 : [DEBUG] (Thread-73 ) http://pi4:81/cmd/up ( POST ): ((), {'command': 'up'}) | {}
2020-04-25 13:20:23,588 : [INFO] (Thread-73 ) processing Command "up" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-25 13:20:23,589 : [DEBUG] (Thread-73 ) rise shutter "0x279623"
2020-04-25 13:20:23,589 : [INFO] (Thread-73 ) [Lounge Blind] Going up
2020-04-25 13:20:23,590 : [DEBUG] (Thread-73 ) sendCommand: Waiting for Lock
2020-04-25 13:20:23,590 : [DEBUG] (Thread-73 ) sendCommand: Lock aquired
2020-04-25 13:20:23,592 : [DEBUG] (Thread-73 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-25 13:20:23,600 : [INFO] (Thread-73 ) Remote  :      0x279623 (Lounge Blind)
2020-04-25 13:20:23,601 : [INFO] (Thread-73 ) Button  :      0x02
2020-04-25 13:20:23,601 : [INFO] (Thread-73 ) Rolling code : 4062
2020-04-25 13:20:23,601 : [INFO] (Thread-73 ) 
2020-04-25 13:20:23,602 : [INFO] (Thread-73 ) Frame  :    0xA7 0x20 0x0F 0xDE 0x27 0x96 0x23 
2020-04-25 13:20:23,602 : [INFO] (Thread-73 ) With cks  : 0xA7 0x28 0x0F 0xDE 0x27 0x96 0x23 
2020-04-25 13:20:23,603 : [INFO] (Thread-73 ) Obfuscated :0xA7 0x8F 0x80 0x5E 0x79 0xEF 0xCC 
2020-04-25 13:20:24,217 : [DEBUG] (Thread-73 ) sendCommand: Lock released
2020-04-25 13:20:24,221 : [DEBUG] (Thread-75 ) [Lounge Blind] Waiting for operation to complete for 27.0 seconds
2020-04-25 13:20:41,121 : [DEBUG] (Thread-76 ) http://pi4:81/cmd/stop ( POST ): ((), {'command': 'stop'}) | {}
2020-04-25 13:20:41,122 : [INFO] (Thread-76 ) processing Command "stop" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-25 13:20:41,123 : [DEBUG] (Thread-76 ) stop shutter "0x279623"
2020-04-25 13:20:41,123 : [INFO] (Thread-76 ) [Lounge Blind] Stopping
2020-04-25 13:20:41,123 : [DEBUG] (Thread-76 ) sendCommand: Waiting for Lock
2020-04-25 13:20:41,124 : [DEBUG] (Thread-76 ) sendCommand: Lock aquired
2020-04-25 13:20:41,126 : [DEBUG] (Thread-76 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-25 13:20:41,133 : [INFO] (Thread-76 ) Remote  :      0x279623 (Lounge Blind)
2020-04-25 13:20:41,134 : [INFO] (Thread-76 ) Button  :      0x01
2020-04-25 13:20:41,134 : [INFO] (Thread-76 ) Rolling code : 4063
2020-04-25 13:20:41,135 : [INFO] (Thread-76 ) 
2020-04-25 13:20:41,135 : [INFO] (Thread-76 ) Frame  :    0xA7 0x10 0x0F 0xDF 0x27 0x96 0x23 
2020-04-25 13:20:41,136 : [INFO] (Thread-76 ) With cks  : 0xA7 0x1A 0x0F 0xDF 0x27 0x96 0x23 
2020-04-25 13:20:41,136 : [INFO] (Thread-76 ) Obfuscated :0xA7 0xBD 0xB2 0x6D 0x4A 0xDC 0xFF 
2020-04-25 13:20:41,750 : [DEBUG] (Thread-76 ) sendCommand: Lock released
2020-04-25 13:20:41,750 : [DEBUG] (Thread-76 ) [0x279623] Previous position: 0
2020-04-25 13:20:41,750 : [DEBUG] (Thread-76 ) [0x279623] Seconds since last command: 18
2020-04-25 13:20:41,750 : [DEBUG] (Thread-76 ) [0x279623] Duration percentage: 67
2020-04-25 13:20:41,751 : [INFO] (Thread-76 ) Received request to set Shutter 0x279623 to 67
2020-04-25 13:20:41,751 : [INFO] (Thread-76 ) sending message to MQTT: somfy/0x279623/level/set_state = 67
2020-04-25 13:20:51,250 : [DEBUG] (Thread-75 ) [Lounge Blind] Discard final position as position is now: 67
2020-04-25 13:20:52,495 : [DEBUG] (Thread-78 ) http://pi4:81/cmd/up ( POST ): ((), {'command': 'up'}) | {}
2020-04-25 13:20:52,497 : [INFO] (Thread-78 ) processing Command "up" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-25 13:20:52,497 : [DEBUG] (Thread-78 ) rise shutter "0x279623"
2020-04-25 13:20:52,498 : [INFO] (Thread-78 ) [Lounge Blind] Going up
2020-04-25 13:20:52,498 : [DEBUG] (Thread-78 ) sendCommand: Waiting for Lock
2020-04-25 13:20:52,498 : [DEBUG] (Thread-78 ) sendCommand: Lock aquired
2020-04-25 13:20:52,500 : [DEBUG] (Thread-78 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-25 13:20:52,508 : [INFO] (Thread-78 ) Remote  :      0x279623 (Lounge Blind)
2020-04-25 13:20:52,508 : [INFO] (Thread-78 ) Button  :      0x02
2020-04-25 13:20:52,509 : [INFO] (Thread-78 ) Rolling code : 4064
2020-04-25 13:20:52,509 : [INFO] (Thread-78 ) 
2020-04-25 13:20:52,510 : [INFO] (Thread-78 ) Frame  :    0xA7 0x20 0x0F 0xE0 0x27 0x96 0x23 
2020-04-25 13:20:52,510 : [INFO] (Thread-78 ) With cks  : 0xA7 0x25 0x0F 0xE0 0x27 0x96 0x23 
2020-04-25 13:20:52,510 : [INFO] (Thread-78 ) Obfuscated :0xA7 0x82 0x8D 0x6D 0x4A 0xDC 0xFF 
2020-04-25 13:20:53,124 : [DEBUG] (Thread-78 ) sendCommand: Lock released
2020-04-25 13:20:53,125 : [DEBUG] (Thread-80 ) [Lounge Blind] Waiting for operation to complete for 8.91 seconds
2020-04-25 13:21:02,045 : [DEBUG] (Thread-80 ) [Lounge Blind] Set new final position: 100
2020-04-25 13:21:02,046 : [INFO] (Thread-80 ) Received request to set Shutter 0x279623 to 100
2020-04-25 13:21:02,046 : [INFO] (Thread-80 ) sending message to MQTT: somfy/0x279623/level/set_state = 100
[... cut ...]
2020-04-25 13:21:09,386 : [DEBUG] (Thread-81 ) sendCommand: Lock released
2020-04-25 13:21:09,387 : [DEBUG] (Thread-81 ) [0x279623] Previous position: 100
2020-04-25 13:21:09,387 : [DEBUG] (Thread-81 ) [0x279623] Seconds since last command: 16
2020-04-25 13:21:09,387 : [DEBUG] (Thread-81 ) [0x279623] Duration percentage: 59
2020-04-25 13:21:09,387 : [INFO] (Thread-81 ) Received request to set Shutter 0x279623 to 159
2020-04-25 13:21:09,388 : [INFO] (Thread-81 ) sending message to MQTT: somfy/0x279623/level/set_state = 159
`

MichaelB2018, 2020-04-26:
I found another issue and fixed it. Can you check again. 

But keep in mind that when you start up Pi-Somfy, it does not know the real current position of the blind. So you have to rise or lower it first time using Pi-Somfy, and from then onwards it tracks the state until it's restarted (so it does not persist the state).

rbswift, 2020-04-26:
Thanks again. Yes, understand there's no persistence of position but I'm still getting output outside the expected range after updating sorry...

```
2020-04-26 15:24:43,636 : [DEBUG] (Thread-58 ) http://pi4:81/
2020-04-26 15:24:44,293 : [DEBUG] (Thread-88 ) http://pi4:81/cmd/getConfig ( GET ): ((), {'command': 'getConfig'}) | {}
2020-04-26 15:24:44,297 : [INFO] (Thread-88 ) processing Command "getConfig" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([])])
2020-04-26 15:24:44,304 : [DEBUG] (Thread-88 ) getConfig called, sending: {"Latitude": -33.8811357990966, "Longitude": 151.20998382568362, "Schedule": {}, "ShutterDurations": {"0x27
9621": 13, "0x279622": 25, "0x279623": 27, "0x279624": 23}, "Shutters": {"0x279621": "Clerestory", "0x279622": "Back Shutter", "0x279623": "Lounge Blind", "0x279624": "French Doors"
}}
2020-04-26 15:24:51,620 : [DEBUG] (Thread-94 ) http://pi4:81/cmd/stop ( POST ): ((), {'command': 'stop'}) | {}
2020-04-26 15:24:51,627 : [INFO] (Thread-94 ) processing Command "stop" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-26 15:24:51,628 : [DEBUG] (Thread-94 ) stop shutter "0x279623"
2020-04-26 15:24:51,628 : [INFO] (Thread-94 ) [Lounge Blind] Stopping
2020-04-26 15:24:51,629 : [DEBUG] (Thread-94 ) sendCommand: Waiting for Lock
2020-04-26 15:24:51,629 : [DEBUG] (Thread-94 ) sendCommand: Lock aquired
2020-04-26 15:24:51,632 : [DEBUG] (Thread-94 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-26 15:24:51,640 : [INFO] (Thread-94 ) Remote  :      0x279623 (Lounge Blind)
2020-04-26 15:24:51,641 : [INFO] (Thread-94 ) Button  :      0x01
2020-04-26 15:24:51,641 : [INFO] (Thread-94 ) Rolling code : 4098
2020-04-26 15:24:51,641 : [INFO] (Thread-94 ) 
2020-04-26 15:24:51,642 : [INFO] (Thread-94 ) Frame  :    0xA7 0x10 0x10 0x02 0x27 0x96 0x23 
2020-04-26 15:24:51,642 : [INFO] (Thread-94 ) With cks  : 0xA7 0x14 0x10 0x02 0x27 0x96 0x23 
2020-04-26 15:24:51,643 : [INFO] (Thread-94 ) Obfuscated :0xA7 0xB3 0xA3 0xA1 0x86 0x10 0x33 
2020-04-26 15:24:52,323 : [DEBUG] (Thread-94 ) sendCommand: Lock released
2020-04-26 15:24:52,324 : [DEBUG] (Thread-94 ) [0x279623] Previous position: 100
2020-04-26 15:24:52,324 : [DEBUG] (Thread-94 ) [0x279623] Seconds since last command: 23390
2020-04-26 15:24:52,327 : [WARNING] (Thread-94 ) [0x279623] Too much time since last command, unable to compute correct position
2020-04-26 15:24:52,327 : [INFO] (Thread-94 ) Received request to set Shutter 0x279623 to 50
2020-04-26 15:24:52,327 : [INFO] (Thread-94 ) sending message to MQTT: somfy/0x279623/level/set_state = 50
2020-04-26 15:25:03,239 : [DEBUG] (Thread-96 ) http://pi4:81/cmd/up ( POST ): ((), {'command': 'up'}) | {}
2020-04-26 15:25:03,240 : [INFO] (Thread-96 ) processing Command "up" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-26 15:25:03,241 : [DEBUG] (Thread-96 ) rise shutter "0x279623"
2020-04-26 15:25:03,241 : [INFO] (Thread-96 ) [Lounge Blind] Going up
2020-04-26 15:25:03,241 : [DEBUG] (Thread-96 ) sendCommand: Waiting for Lock
2020-04-26 15:25:03,242 : [DEBUG] (Thread-96 ) sendCommand: Lock aquired
2020-04-26 15:25:03,244 : [DEBUG] (Thread-96 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-26 15:25:03,252 : [INFO] (Thread-96 ) Remote  :      0x279623 (Lounge Blind)
2020-04-26 15:25:03,252 : [INFO] (Thread-96 ) Button  :      0x02
2020-04-26 15:25:03,253 : [INFO] (Thread-96 ) Rolling code : 4099
2020-04-26 15:25:03,253 : [INFO] (Thread-96 ) 
2020-04-26 15:25:03,253 : [INFO] (Thread-96 ) Frame  :    0xA7 0x20 0x10 0x03 0x27 0x96 0x23 
2020-04-26 15:25:03,254 : [INFO] (Thread-96 ) With cks  : 0xA7 0x26 0x10 0x03 0x27 0x96 0x23 
2020-04-26 15:25:03,254 : [INFO] (Thread-96 ) Obfuscated :0xA7 0x81 0x91 0x92 0xB5 0x23 0x00 
2020-04-26 15:25:03,868 : [DEBUG] (Thread-96 ) sendCommand: Lock released
2020-04-26 15:25:03,868 : [DEBUG] (Thread-98 ) [Lounge Blind] Waiting for operation to complete for 13.5 seconds
2020-04-26 15:25:17,375 : [DEBUG] (Thread-98 ) [Lounge Blind] Set new final position: 100
2020-04-26 15:25:17,376 : [INFO] (Thread-98 ) Received request to set Shutter 0x279623 to 100
2020-04-26 15:25:17,376 : [INFO] (Thread-98 ) sending message to MQTT: somfy/0x279623/level/set_state = 100
2020-04-26 15:25:20,632 : [DEBUG] (Thread-99 ) http://pi4:81/cmd/stop ( POST ): ((), {'command': 'stop'}) | {}
2020-04-26 15:25:20,634 : [INFO] (Thread-99 ) processing Command "stop" with parameters: CombinedMultiDict([ImmutableMultiDict([]), ImmutableMultiDict([('shutter', '0x279623')])])
2020-04-26 15:25:20,634 : [DEBUG] (Thread-99 ) stop shutter "0x279623"
2020-04-26 15:25:20,635 : [INFO] (Thread-99 ) [Lounge Blind] Stopping
2020-04-26 15:25:20,635 : [DEBUG] (Thread-99 ) sendCommand: Waiting for Lock
2020-04-26 15:25:20,635 : [DEBUG] (Thread-99 ) sendCommand: Lock aquired
2020-04-26 15:25:20,637 : [DEBUG] (Thread-99 ) CONFIG FILE WRITE ->> mySectionStart = 73, mySectionEnd = 77, myLine = 76
2020-04-26 15:25:20,645 : [INFO] (Thread-99 ) Remote  :      0x279623 (Lounge Blind)
2020-04-26 15:25:20,646 : [INFO] (Thread-99 ) Button  :      0x01
2020-04-26 15:25:20,646 : [INFO] (Thread-99 ) Rolling code : 4100
2020-04-26 15:25:20,647 : [INFO] (Thread-99 ) 
2020-04-26 15:25:20,647 : [INFO] (Thread-99 ) Frame  :    0xA7 0x10 0x10 0x04 0x27 0x96 0x23 
[... cut ...]
2020-04-26 15:25:21,261 : [DEBUG] (Thread-99 ) sendCommand: Lock released
2020-04-26 15:25:21,262 : [DEBUG] (Thread-99 ) [0x279623] Previous position: 100
2020-04-26 15:25:21,262 : [DEBUG] (Thread-99 ) [0x279623] Seconds since last command: 17
2020-04-26 15:25:21,262 : [DEBUG] (Thread-99 ) [0x279623] Duration percentage: 63
2020-04-26 15:25:21,262 : [INFO] (Thread-99 ) Received request to set Shutter 0x279623 to 163
2020-04-26 15:25:21,263 : [INFO] (Thread-99 ) sending message to MQTT: somfy/0x279623/level/set_state = 163
```

rbswift, 2020-04-26:
PR added #61 
Seems to work for me. Cheers

MichaelB2018, 2020-04-26:
code merged, see #61

closed 2020-04-26 by MichaelB2018