Nickduino/Pi-Somfy #63: persist shutter position

rbswift, 2020-04-27:
It might be nice to persist each shutter position perhaps in a [ShutterPositions] section of the .conf file similar to how the rolling codes are persisted. This would make the software more reliable for users who restart their raspberry Pi regularly or occasionally. The position could also then be represented in the Web UI.

This is a bit beyond my capabilities unfortunately but perhaps if I get more into python I will revisit this myself if nobody else beats me to it.

MichaelB2018, 2020-05-02:
Thanks for the suggestion. I'd be happy to support a pull request for this. Obviously the other alternative you have is to utilize the MQTT interface and then use an external tool such as Home Assistant to maintain the state of the shutter.

Nickduino, 2021-10-10:
I'm closing that issue for the moment but it is a cool idea indeed. Feel free to reopen it when you're ready to make it happen

closed 2021-10-10 by Nickduino