Nickduino/Pi-Somfy #65: Problems with config file

nstone33, 2020-05-03:
I set up Pi-Somfy a few months ago and it worked fine for a while.  I have now developed problems, namely the two blinds I have do not work even on the web interface, before I even get to Alexa integration (which again did work).  The issue I am facing in trying to go back to the start and set them up again is that if delete the shutters in the web interface, and then  refresh the page they come back.  I have also tried deleting the conf file referred to in the startup command  (home/pi/Pi-Somfy/operateShutters.conf).  On starting up again, a new conf file is created, but the web interface still shows the two shutters it previously listed. It appears to be finding the shutters from another file, as the new conf file shows no shutters listed.. don't understand.

There is clearly something I have missed here.  Any advice on how to resolve this, i.e. delete the old shutters permanently so I can set them up again?

closed 2020-05-07 by nstone33

MichaelB2018, 2020-05-07:
Did you solve the problem?

nstone33, 2020-05-07:
That particular issue appears to have resolved.  But I have now added the
two blinds, they have appeared correctly in the conf file, but I can’t get
them to respond to commands, either using the buttons on the web interface
or by copying the script using the relevant blind name.

I’ve checked the transmitter is still plugged in securely; it is connected
to the same pins it was on the last time the system worked (which was a
long time ago!).

Anything else I can do to check what is going on?

Thanks

Neil