Nickduino/Pi-Somfy #67: Raspberry Pi 1 Compatibility

StarScream159, 2020-05-21:
Hey, I'm trying to get this to work with my old Raspberry Pi 1 (I think it's B?) with the 26 pin header and having some trouble. I don't know if it is my transmitters or the Pi or something else. I found conflicting reports of the pinout for the GPIO, so maybe that?

[This site](https://pi4j.com/1.2/pins/model-b-rev1.html) is different then what I've seen written elsewhere. I went with image from [this page here](https://raspi.tv/2014/rpi-gpio-quick-reference-updated-for-raspberry-pi-b) since that seems to match the other Pi versions and other sites reported the same.

However, it won't work. I can't get the blind to accept the programming/virtual remote with my real remotes. I place the blinds into programming mode (using the original working remotes by holding the back little pinhole button down for a few seconds until the target bind "jogs"). Then I click the "try programming again" button but nothing happens.

I don't see any errors in the log.
I tried all 5 of the transmitter boards I made.
I tried different GPIO pins (and changed them in the configuration).
I tired moving the Pi very close to the blinds (in case the antenna isn't working).

But nothing. I can't imagine all 5 of the transmitters are broken. I used the exact components listed here (including the optional antenna).

Anyone got any ideas? I have 3 blinds to add.

rbswift, 2020-05-22:
Suggestions:

Do a full reset on a motor (on, 2sec off, 10 sec on, 2 sec off, on) in case you have reached the limit on the number of remotes (this varies by motor)

Older raspberry pis have a shorter 26 pin header but that should be fine. Use the official pinout here: https://www.raspberrypi.org/documentation/usage/gpio/README.md

Note: the numbering of the GPIO pins is not in numerical order.

You could try a simple LED test to cofirm your GPIO is working: https://www.raspberrypi.org/blog/how-to-use-an-led-with-raspberry-pi/

If you really want to confirm the radio transmitter you could set up a basic SDR receiver to monitor 433.42MHz but that's a bit more work.

[Suggestions if anyone has some free time - Does an LED illuminate briefly if connected to the relevant GPIO pin when Pi-Somfy is used? That might be a good troubleshooting step but I've never tried it. Also, a quick and dirty SDR guide using something like the Nooelec NESDR Mini 2 SDR could be useful. I bought one a while back but never got around to setting it up :) ]

StarScream159, 2020-05-24:
Thanks for the reply. Still having issues though.

I tested the theory that the motors were at the remote limit by pairing a Telis 4 Somfy that had a free channel to one of the motors. The Telis 4 paired fine, and then I removed it fine. So the motor was able to take a new remote no problem.

I retested my 5 transmitters and attempted to pair but same outcome as above. I can't believe all 5 are bad,

I wish I had a way to test my transmitters without spending $50 on a usb SDR receiver (shipping in my area is the same price as the receiver itself). But without it there is no way to verify that my hardware is working. It could be the Pi or the transmitters.

rbswift, 2020-05-24:
Maybe post your operameshutter.log file and someone can see if there is anything obvious there? Did you try an LED on your GPIO to confirm that you are using the right pin and it is working?

Nickduino, 2020-05-24:
> I wish I had a way to test my transmitters without spending $50 on a usb SDR receiver (shipping in my area is the same price as the receiver itself). But without it there is no way to verify that my hardware is working. It could be the Pi or the transmitters.

You could write a sketch or a script to receive signals of an Arduino or RPi

StarScream159, 2020-05-24:
I could. But it would be cheaper for me to go the SDR route. To test the transmitter with an Adrunio or RPi I would need to buy another Pi or an Adrunio (and eveyrting to get it working: power supply, storage card, etc.), and also buy a receiver module for the correct frequency.

I'll look into the LED test. I don't think I have the required components kicking around so that's something I'll have to order/buy.

Nickduino, 2020-05-24:
Your emitter didn't come with a receiver add well? The $1 Chinese stuff often come in pairs
(These aren't so picky about frequency: if you actually emit around 433 MHz, they should pick it up)

A Chinese Arduino nano/micro/mini clone is what, $2-3 shipped?

Have you checked your wiring 15 times? Changed the output pin? Got a somewhat decent antenna ? Made sure your oscillator is the right frequency?

StarScream159, 2020-05-29:
Finally got around to testing this. Yes, you are right in that the transmitter came with a receiver. I discounted it because I assumed it would be on the incorrect frequency now. But since you said it should still pick it up I gave it a shot.

I found the package [rpi-rf](https://github.com/milaq/rpi-rf) to do my tests with. I triple checked my pins and fired up two shells (one to run the receiver script and one to do the sending). Unfortunately the tests are inconclusive. I saw some random data coming through though:
```pi@raspberrypi:~ $ rpi-rf_receive
2020-05-29 03:41:49 - [INFO] rpi-rf_receive: Listening for codes on GPIO 27
2020-05-29 03:42:05 - [INFO] rpi-rf_receive: 1 [pulselength 1111, protocol 2]
2020-05-29 03:43:05 - [INFO] rpi-rf_receive: 2 [pulselength 1258, protocol 4]
2020-05-29 03:43:43 - [INFO] rpi-rf_receive: 8 [pulselength 1035, protocol 2]
2020-05-29 03:44:00 - [INFO] rpi-rf_receive: 4 [pulselength 713, protocol 5]
2020-05-29 03:45:31 - [INFO] rpi-rf_receive: 8 [pulselength 324, protocol 1]
2020-05-29 03:46:01 - [INFO] rpi-rf_receive: 262144 [pulselength 1602, protocol 4]
2020-05-29 03:46:10 - [INFO] rpi-rf_receive: 64 [pulselength 1275, protocol 4]
2020-05-29 03:46:47 - [INFO] rpi-rf_receive: 68 [pulselength 794, protocol 2]
2020-05-29 03:47:00 - [INFO] rpi-rf_receive: 131072 [pulselength 2209, protocol 4]
2020-05-29 03:49:31 - [INFO] rpi-rf_receive: 8 [pulselength 520, protocol 5]
```
But that wasn't from me, I don't think. I expected to see the same decimal value I sent (116) on the receiver. The data above I would get at random times (when not sending but also sometimes while sending). However it never matched my sending value. Unless I'm not understand how this would work, I don't think the data that was being received was from the transmitter.

So still not sure if the receiver is too picky now that the transmitter and receiver are on different frequency and that's why this test didn't work, or if it is my Pi or if the transmitter is broken (all 5 of them). If it has received data then at least I'd know it was sending data properly.

> Have you checked your wiring 15 times? Changed the output pin? Got a somewhat decent antenna ? Made sure your oscillator is the right frequency?
Yes. Checked it multiple times. Over and over. I've tried 4 different output pins. I purchased the same antenna as linked in the readme. And I purchased the same oscillator as in the readme. My exact purchased hardware:
[transmitter and receiver kit Module](https://www.ebay.ca/itm/5x-433Mhz-RF-transmitter-and-receiver-kit-Module-Arduino-ARM-WL-MCU-Raspberry-Nz/293129008524?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
[Antenna](https://www.ebay.ca/itm/5PCS-433MHZ-Helical-Antenna-for-Arduino-Remote-Control/311598741959?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
[saw resonator crystals](https://www.ebay.ca/itm/5pcs-433-42m-433-42mhz-r433-f433-saw-resonator-crystals-to-39-new/232574365405?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
I also got the wires that were optional, the exact link I bought is now dead, but [this looks like them](https://www.ebay.ca/itm/40PCS-Dupont-wire-20cm-Cables-Line-Jumper-1p-1p-pin-Connector-Female-to-Female/221480809935?hash=item339148f5cf:g:JwgAAOSw1DtXKXFk).

I guess if I want to test further I'll have to order either the LED components or an Arduino (but since I tested the receiver with my current Pi not sure if an Ardunio would tell me anything new). Or get the SDR.

Thanks for the help.

StarScream159, 2020-05-29:
> Maybe post your operameshutter.log file and someone can see if there is anything obvious there? Did you try an LED on your GPIO to confirm that you are using the right pin and it is working?

Sorry missed this. I cleared my config. Re-ran the first command as in the readme that generated a new one from the default. I then used nano to edit the port and GPIO pin that I have the transmitter connected to. And then ran the script. It started the web interface which I accessed and tried to add a blind. I used my current real remote to put it in programming mode (which was confirmed via the jog it does) and clicked the add button but nothing happened. I then tried I think 5 times to click the "try again" button. Nothing happened.

The log can be [viewed over on pastebin](https://pastebin.com/LCTAL3CX), since it is a little long.

Nickduino, 2020-05-29:
1. Not sure the same Pi can emit and receive at the same time. Maybe, maybe not.
2. rpi-rf uses its own protocole. There is no way it would decipher Somfy's specific one and tell you the remote number and rolling code: Somfy's code is obfuscated (pretty badly done but still obfuscated)
3. So you soldered your 433.42 MHz oscillators on the **five** emitters? Why?
4. Where did you solder the antenna ? Picure?

StarScream159, 2020-05-29:
1. Yeah that would be another issue then, however I'd be surprised since you can communicate with multiple pins at the same time - so why wouldn't you be able to?
2. I used rpi-rf to do the sending and receiving. There are two scripts included. I didn't use Pi-Somfy at all for this test.
3. Haha yeah - all five. That was a mistake. Should have left 1 or 2 alone. But I was doing it and I know 1 of them is messed up, so while I was on a roll doing it and had all the stuff out I figured I'd do the rest so that way I'd have **at least** 1 or 2 working that I could use. Or so I thought.
4. Yeah here is a few pictures of them. You can view them over on this[ imgur album](https://imgur.com/a/vxiYzYy). The fifth one is attached to the Pi currently, that's why it isn't in the picture. But it is the same as the others. The one with the "x" is the known bad one.

I found a SDR that is on Amazon that might work (posted about it issue #49). With that I can just add it to my next order to save on the shipping. If there is no other things to try to test, I might just put this little project on hold until I get that.

Really appreciate the continued replies.

Nickduino, 2020-05-29:
1. No idea, I was just wondering if that might throw the timing off or something.
2. Do they say on that other repo the sending and receiving should work on the same Pi? I'm 99.999999999999% positive your receiver should receive at 433.42: it's a shitty receiver (I've got some of them) that accepts a pretty large frequency range around 433 MHz
4. Looks alright. I was asking mostly because someone else had soldered his antenna in lieu of an inductor, easy mistake.

> Really appreciate the continued replies.

Sadly, there's not much I can do to help you. I'll give you two more leads, though.
5. On this topic, https://forum.arduino.cc/index.php?topic=208346.90, you should find an Arduino sketch that should be able to decode a Somfy remote (or one you emulate with the Pi)
6. My first emitter wasn't a chinese thing with a replaced oscillator but my remote itself: I powered it from my Arduino and sent the data to [this pin](https://lh3.googleusercontent.com/-6kNyMMGRxXg/VlYn_ZMBk_I/AAAAAAAAN7s/X3B9l7eSEH8/s720-Ic42/MacroPhoto%2525201.png). With luck, you may be able to identify the same pin on your remote (which very likely will have a different MCU as mine aren't actually Somfy) and test with it? That may be risky for the remote, though, and I can't advise it.

Lastly to cheer you up: I understand these steps are not very pleasant and all but imagine how frustrating things were a few years ago when I was experimenting with all that, not even knowing if it would be at all possible, not being able to identify if my issues were hardware or software...

StarScream159, 2020-10-08:
Thought I'd update this issue. A few months ago I ended up buying a SDR off Amazon and testing my transmitters. They failed. All of them. I tested the SDR with the remote and it works 100%. I can see the signal coming in. So I know the SDR works and I am using it right. So that means my transmitters, Raspberry Pi, or cables are bad. I saw a helpful post from @kjellhult over on issue #49 that I think is my problem.

When I replaced the crystal on the board I soldered it flush to the board. I didn't know it needed a gap or the case of the crystal would short out to a pin - maybe that should be in the readme. I'm going to, eventually, try to fix it and resolder it all. I'll update here when I try the next step.

Nickduino, 2020-11-04:
> I'll update here when I try the next step.

Did you solve it?

StarScream159, 2021-01-09:
An update on this. I haven't gotten any anywhere on this due to my soldering skills being crapping and needing help. I originally did this with a friend and due to the pandemic haven't been able to get together to try again. Maybe sometime this year. I'll update when I do.

alonjr, 2021-08-01:
@StarScream159 any news? I'm dying to know the end of your thrilling adventures 😂. I too am having issues with getting the hardware to work.

closed 2021-10-10 by Nickduino