Nickduino/Pi-Somfy #71: MQTT and Alexa integration not launching

Wolbyworld, 2020-06-01:
Hey,
Launching the service from crontab the MQTT nor Alexa integrations are working. When running manually the commend, there is an error on myconfig.py:163 section ShutterIntermediatePositions entity 0x279621.

I have tried to debug it but can´t go further than there. 

I have not modify anything from the current version other than adding a shutter via web portal (which I can perfectly control from there)

Thanks

Wolbyworld, 2020-06-04:
Alexa is now working, small tweak on my network settings. Can´t manage to make work the MQTT and HA integration. My current hypothesis is that is related to my MQTT network not using any user and password. Could that be the reason?

Wolbyworld, 2020-06-04:
Solved. If no password on MQTT, comment line 101 on mymqtt.py

MichaelB2018, 2020-06-07:
fixed. just pull the latest version

closed 2020-06-07 by MichaelB2018