Nickduino/Pi-Somfy #73: WebGUI not starting ([Errno 98] Address already in use)

snorro80, 2020-06-12:
Installed Pi-somfy on an RPI with latest Buster following the guide.
After installation webGUI can't be reached. 
When manually starting the webserver i get the following error:
`[Errno 98] Address already in use`

Tried other port, eg 8082, no error on manual start but webGUI still can't be reached.

firewiremb, 2020-07-05:
If you follow the commissioning instructions step by step (alternative 5 systemd service), the file operateShutters.conf is created in the directory /home/pi. Apparently this is not used.
The file with the same name in the path /home/pi/Pi-Somfy is used. To change the port for the web interface you have to edit here.

Since port 80 is already used on my Pi, I use port 10055.

Nickduino, 2020-07-09:
Was it solved @snorro80 ?

snorro80, 2020-07-10:
Haven't been able to try it due to holiday. 
Will let you know if it worked as soon as I tried your solution.