Nickduino/Pi-Somfy #76: Blinds stop responding after setup

githubcoetseeuk, 2020-06-24:
I just wanted to check if anybody else has had this situation...? I've now tried on multiple Somfy RTS 28 wirefree blinds - the initial setup using the web interface to add a blind works by using the Telis 1 remote and the Add/Remove function on the web interface (Program button) - the blind jogs to show it's accepted the signal sent by the web interface. After that the blind will respond to the up/down/stop command from the web interface for about 10 seconds - then won't anymore. You have to go through the blind programming sequence of removing and adding the web interface again using the Telis remote and Program button - which results in the same 10 second window of the blind responding and then not anymore? Any ideas?

rbswift, 2020-06-24:
Yep. battery motors go to sleep and seem to need a bit more aggressive RF commands to wake them up. Try setting SendRepeat = 3 in your operateShutters.conf

githubcoetseeuk, 2020-06-24:
Thanks for the quick help - and I suspect you might be right - but I'm afraid it doesn't fix the problem - even after increasing SendRepeat to 10. The reason I suspect you're right - if I use the Telis remote just to move the blind a fraction and then stop it using the remote - the web interface works again for about 10 seconds - so I assume the remote wakes up the blind which then responds to the web interface. Could the battery motors be expecting a different 'wake-up' command to the wired motors?

rbswift, 2020-06-24:
Interesting... I never quite figured out what causes the battery motors to behave differently. I mostly fixed it with SendRepeat = 3 (I think more than 3 is actually a press and hold which may not work or may work differently). Does it not work even if you use the web interface repeatedly? Maybe that's a different problem. I think we need someone to fire up an SDR and determine what's different between a genuine Somfy remote and the Pi-Somfy project RF transmitter. I don't really know enough about SDR sadly but am happy to try and help

githubcoetseeuk, 2020-06-24:
Nope - you can use the web interface as much as you like - no response after initial setup - which again would mean that the motor was woken up by using the remote to put the blind into programming mode during setup... I would also assume that if you just keep pressing the web interface buttons without the blind responding - eventually the rolling code is going to go out of sync. The web interface works every time if you first wake the blind using the remote. PS - Pi-Somfy transmitter is right next to the blind and has an antenna - so signal strength shouldn't be a problem.

Nickduino, 2020-06-24:
> I think we need someone to fire up an SDR

Or just an oscilloscope? And check the signal on the antenna pin

githubcoetseeuk, 2020-06-25:
Ordered SDR - will let you know what I find...

Nickduino, 2020-06-25:
@githubcoetseeuk  Nice, I'm impatient to see what you discover

yavinmaster, 2021-01-31:
@githubcoetseeuk did you ever get this sorted.  About to install these blinds, still waiting on my crystal to be delivered, and just checking what problems I might run in to.

yavinmaster, 2021-02-14:
I have my setup operational now and can report that I don't appear to have any issues.
Great work with this software.

Nickduino, 2021-02-14:
Awesome!

@githubcoetseeuk Feel free to reopen if you want to post your findings. For the moment, though...

closed 2021-02-14 by Nickduino