Nickduino/Pi-Somfy #78: Partial Operation not working :/

xChr11s, 2020-07-01:
Hello together,

I installed everything today on my raspberry pi3 and could connect my somfy instantly.
The manual Operation is working fine so far. 
I tested a bit around with the Scheduled Operations and figured out, that the shutter is either completely down or up.
I tried setting 50%, but it went completely down.
Any Idea to debug or fix this?

May I did something wrong? :/

Kind regards

rbswift, 2020-07-02:
you probably need to put the number of seconds it takes your shutter to open/close in that section

xChr11s, 2020-07-02:
> you probably need to put the number of seconds it takes your shutter to open/close in that section

Thats a good point. I took the time it needs to go completely down and wrote 26 seconds into the settings.
Tested it again with the shutter completely up and wanted it to go down 70% but it went gone completely.
:/

MichaelB2018, 2020-07-02:
Can you try with 20%. Some slides go down much faster than they go up or vise versa. So maybe that coudl be an issue here,

xChr11s, 2020-07-03:
> Can you try with 20%. Some slides go down much faster than they go up or vise versa. So maybe that coudl be an issue here,

Thanks for your answer. Tested with 20% but it went down completely again :/

Secarius, 2020-07-06:
I have the same issue, partioal operation alwas operates the full close and open. It dosent matter wich operation time i set.
Is there a fix for this?

stalakerob, 2020-08-05:
Hi, I'm interested in that too. Any progress in the meantime on how to do that? 
Thx

firewiremb, 2020-08-05:
Hi, I think I found one of the problems.

An example:

Shutter A is fully opened in the morning an closed for 60% by scheduled operations lets say at 8:00. Shutter A should again be closed by scheduled operations for 60% at 13:00. Between 8:00 and 13:00 the shutter is operated manually.

What happens?

Lets say Shutter A is moved manually to position 40% between 8:00 and 12:00 --> At 13:00 nothing happens. You will find an error in the logfile

Lets say Shutter A is moved manually to position 80% between 8:00 and 12:00 --> At 13:00 Shutter A is moved to positon 60%. Everything is fine

Lets say Shutter A is not moved manually between 8:00 and 12:00 at all --> At 13:00 Shutter A is fully closed.

I changed the code in my installation and the Problem is gone. I tried to created my first pull request. Hopefully it worked. Never done this before.

rbswift, 2020-08-05:
Before making substantial changes to the code may I suggest reflecting each shutter position open/close% state in the web UI? This will help greatly with troubleshooting and understanding when and if the Pi-Somfy state does not correspond with reality.

Currently it's very likely that there will be unpredictable results if the shutter is controlled manually outside Pi-Somfy because  the code has no way of tracking manual movement.

FWIW, I've got a 3 blinds and 2 shutters all running with Pi-Somfy and have found the best option is to put the manual controllers away somewhere safe for emergency use only and do everything either with schedules, scenes or manually on phones. I use Siri/Apple Home > Homebridge > MQTT > Pi-Somfy and it works flawlessly for open/close and partial states.

stalakerob, 2020-08-06:
@firewiremb and @rbswift, thanks for taking the time to reply. I think rbswift hits the point. Since the 433 MHz system is only unidirectional, you cannot find out the current position if it has been changed with a manual controller. 
What you can do though if you want to move the shutter to a specific position using a script or automation and you want to make sure that the position is correct: 
1. first close the shutter with Pi-Somfy (even if it is closed already ). Pi-Somfy now knows the current position is 100
2. move the shutter to the position you want

firewiremb, 2020-08-06:
@rbswift and @stalakerob 
I do not use a controller outside Pi-Somfy. "Mannually" in my post from yesterday means I am using the Pi-Somfy web interface and not a scheduled operation.
I know the system is unidirectional and I made sure that the starting positions of the shutters match those in Pi-Somfy.

My message from yesterday is a response to the start message from xChr11s
The problems described from xChrs11s occur precisely in this configuration I have described yesterday and can be reproduced at any time. Just try.

xChr11s, 2020-09-16:
Hello guys,
Sorry for the long time not answering, but I'm back now. 
Sadly my Raspi SD Card died and I had to get a new one and setup it correctly.
I have a fresh installed Pi Somfy now and stil the same issue with it completely running down when sat at 70% and completely up before. Any Ideas how to troubleshoot this more?

Thanks