Nickduino/Pi-Somfy #80: Error Message in Log File

firewiremb, 2020-07-05:
Thank you very much for this project. Great what was done here.

At first glance, everything works for me as I imagined. However, errors are recorded in the log file. Can someone tell me what the problem is here?

Log file:

2020-07-05 21:20:05,485 : [INFO] (Scheduler ) Send action "down" to shutterId "0x279622" at 2020/07/05 21:20:05
2020-07-05 21:20:05,486 : [INFO] (Scheduler ) [Bad] Going down
2020-07-05 21:20:05,487 : [DEBUG] (Scheduler ) sendCommand: Waiting for Lock
2020-07-05 21:20:05,487 : [DEBUG] (Scheduler ) sendCommand: Lock aquired
2020-07-05 21:20:05,491 : [DEBUG] (Scheduler ) CONFIG FILE WRITE ->> mySectionStart = 105, mySectionEnd = 115, myLine = 107
2020-07-05 21:20:05,506 : [INFO] (Scheduler ) Remote  :      0x279622 (Bad)
2020-07-05 21:20:05,507 : [INFO] (Scheduler ) Button  :      0x04
2020-07-05 21:20:05,507 : [INFO] (Scheduler ) Rolling code : 5
2020-07-05 21:20:05,508 : [INFO] (Scheduler ) 
2020-07-05 21:20:05,508 : [INFO] (Scheduler ) Frame  :    0xA7 0x40 0x00 0x05 0x27 0x96 0x22 
2020-07-05 21:20:05,509 : [INFO] (Scheduler ) With cks  : 0xA7 0x46 0x00 0x05 0x27 0x96 0x22 
2020-07-05 21:20:05,510 : [INFO] (Scheduler ) Obfuscated :0xA7 0xE1 0xE1 0xE4 0xC3 0x55 0x77 
2020-07-05 21:20:06,126 : [DEBUG] (Scheduler ) sendCommand: Lock released
2020-07-05 21:20:06,128 : [ERROR] (Scheduler ) Error: cannot open 0x279622
2020-07-05 21:20:06,129 : [ERROR] (Scheduler ) Traceback (most recent call last):
  File "/home/pi/Pi-Somfy/myscheduler.py", line 341, in run
    self.shutter.lower(shutterId)
  File "/home/pi/Pi-Somfy/operateShutters.py", line 109, in lower
    timeToWait = state.position/100*self.config.Shutters[shutterId]['duration']
TypeError: can't multiply sequence by non-int of type 'float'

lfliess, 2020-08-17:
@FraBoCH the bug is in your MQTT code.

firewiremb, 2020-08-17:
@lfliess I guess you posted in the wrong thread?

lfliess, 2020-08-18:
@firewiremb  nope, the python exception you're showing is a result of a bug added by @FraBoCH 
@Nickduino  - you might want to revert until fixed

FraBoCH, 2020-08-18:
Hi @firewiremb, thanks for reporting this issue.
It seems my code is indeed not very robust when dealing with partial closing. But apart from the errors in the logs, do you observed any strange behavior ?