Nickduino/Pi-Somfy #81: [Question] will my existing controllers still works after using Pi-Somfy

lepirlouit, 2020-07-15:
Before using Pi-Somfy, I don't want to screw up a working installation.
Could you clarify the step of adding a shutter.
What should I do with the shutter ?
What should I do with the existing remote ?

Kind Regards,

lepirlouit, 2020-07-16:
I'll answer to myself :
Yes, existing controllers will still works.

Program procédure :
1. take the working remote of the shutter you want to controle with Pi-Somfy
2. press the program button for about 3 seconds
the shutter should goes up and down briefly
3. press the program button in the Pi-Somfy web interface, next to the remote you want to link to the shutter

Nickduino, 2020-07-16:
Glad you worked it out by yourself

closed 2020-07-16 by Nickduino