Nickduino/Pi-Somfy #82: (Question): Multiple channels are supported?

joselcaguilar, 2020-07-30:
Thanks a bunch for this amazing project! In my case, I have a Somfy Telis 4 RTS Pure remote controller and 2 different blinds, so my idea is control blind 1, blind 2 or both but I have only 1 prog button in my RC, so maybe changing the device id or room in command line could be possible based on [this](https://github.com/sehaas/yfmos) project and issues opened where they are using Sonoff RF Bridge + Tasmota + Portisch firmware.

Thanks in advance for your support!

Nickduino, 2020-07-30:
In normal operations, do you select the channel on the remote and then press up/down to move one specific blind?

If so I can only assume it would behave the same with the program button. I.e. select the channel, press program, make sure only one blind is moving up and down, and the register your new Pi-Somfy remote

joselcaguilar, 2020-07-30:
> In normal operation, do you select the channel on the remote and then press up/down to move one specific blind?
> 
> If so I can only assume it would behave the same with the program button: select the channel, press program, make sure only one blind is moving up and down, and the register your new Pi-Somfy remote

You are right, in normal operation I select the channel on the remote and then press up/down/my(stop), if all lights are blinking, all blinds will be moved at the same time, so I can choose between 4 channels independently or all of them.

Thank you so much for the idea @Nickduino!
I'll test it asap and inform you if everything work as expected when I have the oscillator which I'm waiting to be delivered.

lepirlouit, 2020-07-30:
@joselcaguilar your controller is actually a 5 channel controller 
when all lights are blinking, you just selected the 5th channel.
your blinds are just connected to two different channels.
B1 linked to C1 and C5
B2 linked to C2 and C5

In Pi-Somfy you can add a third remote that will be registered to both blinds

from the docs, a blind can have up to 9 controllers attached to it.

@Nickduino thanks a lot for your project, it works on the first try