Nickduino/Pi-Somfy #83: Somfy RF reader

geokscott, 2020-08-01:
How hard would it be to set up a receiver that could decode remote commands?  It would be nice to make it two-way so a true integration with Home Assistant could be developed.

lepirlouit, 2020-08-03:
I would be interested by this too.
I already have the hardware receiver https://www.amazon.fr/433Mhz-Wireless-Transmitter-Receiver-Arduino/dp/B00IOB9CWE
How to specificly listen on 433.42mhz
Also we need to separate the concept of "blind/shutter" and "remote controller"

Nickduino, 2020-08-03:
TD;LR: I don't have a use for it (nope, not even "just for fun") so I won't do it. But you can.

Receiver frequency probably won't be an issue: the Chinese receivers aren't that picky. So you have to capture the frame and reverse the obfuscation procedure to know what command was sent to what shutter. Everything you need is in the code or the Pushstack article, that's probably still a few days of work but it's very doable.

lepirlouit, 2020-08-04:
what do you mean with the Pushstack article ?

Nickduino, 2020-08-04:
This:

[image]

closed 2021-03-08 by Nickduino