Nickduino/Pi-Somfy #84: Not able to start PIGPIO

itayraviv, 2020-08-04:
Hi,
I followed the instructions and reached to this step:
"Next, let's test if it all works. Start <operateShutters.py> by typing:
sudo python3 /home/pi/Pi-Somfy/operateShutters.py
You should see the help text explaining the Command Line Interface"

and I get this error:
pi@volumio:~/Pi-Somfy$ sudo ./operateShutters.py
[sudo] password for pi:
Creating new config file : /home/pi/Pi-Somfy/operateShutters.conf
/home/pi/Pi-Somfy/defaultConfig.conf
**Not able to start PIGPIO**
pi@volumio:~/Pi-Somfy$

I'm running on a RaspberryPi which runs Volumio that is based on Raspbian (Actually Debian)
I installed all the pre steps according to the guide.

any idea why its not working?

MichaelB2018, 2020-08-04:
Have you done: 

`sudo apt-get install pigpio python-pigpio python3-pigpio
`

Also, what happens when you type:

`sudo pigpiod -l -m
`

itayraviv, 2020-08-05:
yes i installed gpio and python-gpio:

```
pi@volumio:~$ sudo apt-get install pigpio python-pigpio python3-pigpio
Reading package lists... Done
Building dependency tree
Reading state information... Done
pigpio is already the newest version.
python-pigpio is already the newest version.
python3-pigpio is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 24 not upgraded.
pi@volumio:~$

```

when i type what you suggested i get:

```
pi@volumio:~$ sudo pigpiod -l -m
[sudo] password for pi:
pigpiod: invalid option -- 'm'

pigpio V60
Usage: sudo pigpiod [OPTION] ...
   -a value,   DMA mode, 0=AUTO, 1=PMAP, 2=MBOX,  default AUTO
   -b value,   sample buffer size in ms,          default 120
   -c value,   library internal settings,         default 0
   -d value,   primary DMA channel, 0-14,         default 14
   -e value,   secondary DMA channel, 0-14,       default 6
   -f,         disable fifo interface,            default enabled
   -g,         run in foreground (do not fork),   default disabled
   -k,         disable socket interface,          default enabled
   -l,         localhost socket only              default local+remote
   -n IP addr, allow address, name or dotted,     default allow all
   -p value,   socket port, 1024-32000,           default 8888
   -s value,   sample rate, 1, 2, 4, 5, 8, or 10, default 5
   -t value,   clock peripheral, 0=PWM 1=PCM,     default PCM
   -v, -V,     display pigpio version and exit
   -x mask,    GPIO which may be updated,         default board GPIO
EXAMPLE
sudo pigpiod -s 2 -b 200 -f
  Set a sample rate of 2 microseconds with a 200 millisecond
  buffer.  Disable the fifo interface.

pi@volumio:~$
```

rbswift, 2020-08-06:
That looks like a very old version of pigpiod. You've got V60. Latest is V77 and raspbian distro is V71 I think.
Maybe pick a distro with a more recent version or install it manually.
You could also remove the -m option and it should still work but will use about 10% CPU unnecessarily.

lfliess, 2020-08-19:
hey @MichaelB2018 why is -m used for this project? couldn't figure out what it does exactly but I suspect it's giving me trouble when integrating a receiver

rbswift, 2020-08-19:
as http://abyz.me.uk/rpi/pigpio/pigpiod.html

-m disables sampling which is not required for this project because we are not reading from GPIO, only sending. Sampling uses about 10% of the CPU unnecessarily on a raspberry pi but I think that's the only harm in not specifying -m

lfliess, 2020-08-19:
ah... so this is why I can't sample from it properly :)

itayraviv, 2020-08-27:
ok got it installed, i see the webgui but can't make the motor jitter when adding a motor

engelrolling, 2020-09-14:
@itayraviv How did you manage to install the new version ? I upgraded everything ( even `rpi-upate` ) and still the latest version I am able to find of pigpio is v60

Nickduino, 2021-09-22:
> ok got it installed, i see the webgui but can't make the motor jitter when adding a motor

Is your motor Inn programming mode?

closed 2021-09-22 by Nickduino

itayraviv, 2022-05-10:
trying again on this old thread... 
looks like "gpio" command works ("gpio readall" returns the gpio status table) , but pigpio is not a known command... 
and im guessing that the code is trying to use pigpio?
any resolution?

rbswift, 2022-05-10:
pigpio is the name of a library package, not an executable binary. If `sudo apt-get install pigpio` was successful then it is installed.

bclaesen90, 2022-06-03:
After having to change the IP of the Raspi I had the same issue, turned out I just needed to restart the piggiod services to fix the issue.
 `sudo systemctl restart pigpiod`