Nickduino/Pi-Somfy #86: Dead link to eBay "fancy" antenna

numericOverflow, 2020-08-07:
Your eBay link to the "fancy" antenna is dead

`Literally any 17 cm solid core copper wire will do the job. But if you prefer to order a fancy full one, this one will come handy
Order`

Maybe just a picture or description of what you used would be good to avoid dead links in the future?

Nickduino, 2020-08-07:
Hum...it's working for me

[image]

But again you really don't need a finely tuned antenna, the shutters don't care

numericOverflow, 2020-08-07:
Yeah, weird - now it's working for me too.  When I tried it a day or two ago, eBay was throwing some generic "auction can't be found" error.  I just figured too much time elapsed and they purged the listing since the other links had worked OK. 

Oh well, must have been user/device error. Thanks for checking

closed 2020-08-07 by numericOverflow