Nickduino/Pi-Somfy #87: Pi-Somfy Home Assistant Addon

lfliess, 2020-08-08:
Hi,
starting work on a fork on this project that ports it as an addon to home assistant.
the basic idea is that everyone who uses Rpi might as well install home assistant on the device, and have this working on top. Part of the plan is to remove the scheduling layer, since Home Assistant has scheduling and automations.

Anyone other than me wants to give this a run as beta tester?

rbswift, 2020-08-08:
Sounds interesting but HA is quite heavy for simple use cases. For example, I just use mqtt between Pi-Somfy and homebridge and get all the benefits of apple home, siri, scheduling etc. plus the web UI is there if ever needed for emergencies.

lfliess, 2020-08-08:
yes, but it requires extra hardware.
Personally if I already have the RPi, I might as well use it...
the other target audience is people who already have HA running and have Somfy in their house....

rbswift, 2020-08-08:
I'm not sure what you mean by extra hardware. I run both Pi-Somfy and Homebridge on a single RasPi. I also run tvheadend, pi-hole, teslamate, my hvac, my intruder alarm and a few other bits and pieces there too and try to keep everything quite lean so HA would be massive overkill for me or anyone in the apple ecosystem... but I totally get that is not everyone and some people would probably find an HA add-on useful. I prefer discrete microservices approach so that elements can be upgraded or replaced independently through time and have found mqtt to be an excellent solution to avoid tight coupling of async event driven pub/sub communications.

geokscott, 2020-08-08:
lfliess, I would be willing you help you test.  I'm currently running HA and Pi-Somfy on the same Pi.  For a better experience, I think you would need to develop a receiver also so you can try to keep track of device states when there are multiple sources of control.  For example I'm using Pi-Somfy with my HA now with my outdoor awning.  If the wind sensor or the physical remote are used, HA has no idea.  Short of using this nice inexpensive solution, you might be better off programming to a commercial Somfy Hub device.  There are currently two that I know of.

MichaelB2018, 2020-08-08:
@lfliess: would love to help.

lfliess, 2020-08-09:
@geokscott HA have removed support for running the supervised version in a docker, I was working this way as well. Which to me was a goid reason to start this. 
I am working now on mograting this project  to alpine. Si far it seems the main issue is pigpio and getting it working on alpine

geokscott, 2020-08-09:
@lfliess , I'm running Home Assistant Core in a virtual environment on a Pi, which is still okay.  They put depreciation on hold for some of the older options.  I like having the Rassbarian OS so I can run whatever I like on the Pi along with HA.  I was considering switching to Docker, but I'm going to hold off a while until all the support options settle out.  I've been doing manual since the beginning, so having Supervisor or Add-ins is a nice to have rather than a requirement.

lfliess, 2020-08-18:
@MichaelB2018 @geokscott 
just to update on progress - I just managed to get this running as an addon to HA!
it's got the webserver running and exposed externally (you can pick the port), and MQTT.
just one problem - it doesn't work :)
I mean, everything works, and I can even see GPIO traffic, just not getting the awning to move. so I'm trying to figure out if it's my transmitter or something else.

I had to give up on all scheduling logic.
1. it makes sense- you've got it all in HA (even sunrise/sunset logic)
2. couldn't get the required packages for it running on Alpine (dockers in HA run on Alpine).

I also removed Alexa integration. I'm actually not so sure that was the best idea, I guess I can return it. I just figured out once you have MQTT running and HA, most people will have alexa integration.

geokscott, 2020-08-18:
Mines not working either, LOL!  I just upgraded (changed) my installation.  I switched to community supported "[Installing Home Assistant Supervised on Raspberry Pi OS](https://community.home-assistant.io/t/installing-home-assistant-supervised-on-raspberry-pi-os/201836)".

I think my rolling code is out of sync with my unit.  I have not tried to figure it out yet,  I have other hardware on GPIO pins and they are working....

Anyway, how I have Add-On Store....

lfliess, 2020-08-19:
Try to use the web server to re-add the remote. That's actually the only
reason I'm planning to keep it

lfliess, 2020-10-31:
I actually got everything on the software side working. but something seems off with my transmitter....
it has nothing to do with software, my old Pi which worked with the transmitter now doesn't either.

waiting for a new transmitter + resonator crystal to arrive by mail :(

geokscott, 2021-01-27:
> 
> 
> I actually got everything on the software side working. but something seems off with my transmitter....
> it has nothing to do with software, my old Pi which worked with the transmitter now doesn't either.
> 
> waiting for a new transmitter + resonator crystal to arrive by mail :(

I'm in the same "boat".  Waiting for a new set to arrive from China.  Mine quit working.  I tested with a SDR, no signal.

mattesnrw, 2021-02-04:
> I'm not sure what you mean by extra hardware. I run both Pi-Somfy and Homebridge on a single RasPi. I also run tvheadend, pi-hole, teslamate, my hvac, my intruder alarm and a few other bits and [...]

rbswift had written a bit above that he had installed Pi-Somfy and HomeAssistant on one Pi. 

Is this possible in parallel without any problems... or... by using docker? How can it be implemented? What do you have to keep in mind? What IP address do you have to give Pi-Somfy to the MQTT server if it runs on the same Pi? 

Because if Pi-Somfy can't be installed in parallel with HomeAssistant, I don't see much cost advantage against RFLink and RFXCom. 

I would find an add-on in Homeassistant very useful. 

What is the progress there?

korrix, 2022-01-06:
Hello!

I just created an add-on: https://github.com/korrix/pi-somfy-hassio and it works :), but this is very minimal implementation. Please contact me if you would like to contribute and feel free to modify/use/merge back here.

Thanks for this awesome project!

MichaelB2018, 2022-01-06:
Love it! Thanks. But I have one problem... My HASSIO is too far from my shutter for the signal to reach (actually, it's less than 10 meters, but still!). So I need to run PiSomfy on a separate Raspberry Pi. Would be good to have the ability to have a cluster or similar to maybe even a concept of a "Wifi-2-433 repeater" or so. Basically what I'm thinking is rather than having the 433MHz converter on the same Pi as the controller, we could add a new version of sendCommand that connects over WiFi to another Pi (maybe Pi-Zero) that then issues the actual command. Would anybody else be interested in this?

Alternatively we could think of a Custom Integration for HASSIO that connect to the Raspberry Pi on the same Wifi. I'm keen to hear options what others think.

rbswift, 2022-01-06:
@MichaelB2018 might it be easier to create a HASSIO to MQTT interface (or use an existing one if there is such a thing) and then connect the existing build of Pi-Somfy to the same MQTT broker? That's effectively a Wifi-2-433 bridge using standardised MQTT as the middleware. Repeating the 433 transmissions might have some unexpected challenges because signals overlapping signals could cause interference and the main purpose of the rolling code is to prevent a replay attack.

MichaelB2018, 2022-01-06:
You are probably right and PiSomfy obviously has an MQTT interface with HASSIO discovery already. I was just thinking if I run the actual Web-GUI and scheduler on my HASSIO Pi, I'd need something else to make this a viable solution

rbswift, 2022-01-06:
@MichaelB2018 Yes. My approach has been to use MQTT for day-to-day control via upstream multi-purpose UI and scheduling HA systems (in my case Homebridge and NodeRed but really any MQTT enabled HA system chosen can probably be made to work). This relegates the PiSomfy webUI and scheduler for troubleshooting use only.

MichaelB2018, 2022-01-06:
@rbswift, yeah, that's what I do as well. But the HomeAssitant AddOn made me think about this. Using it as a HASSIO Add-on gives benefits in terms of deployment, monitoring and potentially automatic updates, 3 items we have not built into Pi-Somfy. And it could also reduce actual hardware costs down to just a few $, esp if the repeater would be ESP8266 based or so.

rbswift, 2022-01-07:
There are already a few things like that for esp like https://github.com/dmslabsbr/esphome-somfy but people generally seem to prefer the relative simplicity, familiarity and multi purpose nature of a Linux device over the small cost saving. I expect one day the protocol will also be implemented for Tasmota if it hasn’t already. I couldn’t find it. That seems to have quite a strong following but again we are back to replacing oscillators still rather than using an off she shelf re-flashable device