Nickduino/Pi-Somfy #90: one entry operating two shutters

W11T, 2020-08-25:
First of all thanks a lot for this good work!

Setting it up i messed up some entries. Now they operate two shutters. How can this be fixed?
I removed one of those entries and tried to add it again. But this fails. The new entry doesn't register with this shutter any more.

Any ideas?

btw/
what is the use of "Operation Time" in the Web-GUI?

Nickduino, 2020-08-25:
You lost me. How many entries and working shutters have you got?

Have you tried the pairing procedure with your new entry (ies)?

W11T, 2020-08-25:
In total i have 9 entries. Three of them operate two shutters each. Very likely this is my vault - however.
I deleted one of the faulty entries and tried to set it up again. But the shutter didn't responde to the pairing request. I tried several times.
In the meantime i added another entry and was able to pair it with this shutter. This is working well now. 
This added a new tenth entry in operateShutters.conf.
The previous entry was soft-deleted and it looks to me as if it is kind of "poisoned".

I'd really like to understand the logic in the somfy receivers and how to manage the keys in their memories.

W11T, 2020-08-26:
an update:
As described above i managed to add a new entry and to pair it with the shutter.
Now only a few entries are left operational.

The new entry was assigned a shutter code 0x27962A.
Now in operateShutters.conf i see a lower case and an upper case entry of the code:

[ShutterRollingCodes]
0x279621 = 25
...
0x27962A = 5
0x27962a = 10

Now I first have to delete one entry to get the service starting up.

closed 2021-10-10 by Nickduino